justagist/panda_simulator #4: Error when executing move_robot.py

vrontos, 2020-03-24:
Hello,

I followed all the installation steps and i do the following:

1st shell: `$ roslaunch panda_gazebo panda_world.launch`
2nd shell: `cd ~/catkin_ws/src/panda_simulator/panda_simulator_examples/scripts` and `$ python move_robot.py`.

I get the following error:

```aaaa
Traceback (most recent call last):
  File "move_robot.py", line 4, in <module>
    from franka_core_msgs.msg import JointCommand, RobotState
ImportError: No module named franka_core_msgs.msg 
```

P.S.
 At the beginning i was getting an error when launching the panda_world.launch: 
`[ERROR] [1585070496.558123479, 0.613000000]: Could not load controller 'panda_simulator/joint_state_controller' because controller type 'joint_state_controller/JointStateController' does not exist.
`
now it works because i did the following (is it wrong?): `sudo apt-get install ros-melodic-joint-state-controller
`
Thanks

justagist, 2020-03-24:
It looks like you haven't built the [frank_ros_interface](https://github.com/justagist/franka_ros_interface) package correctly, or is not sourced. The `franka_core_msgs`  package should automatically be installed if you built `franka_ros_interface`. Try rebuilding your catkin workspace after making sure that `franka_ros_interface` is present in your `src` folder along with all other dependent packages (listed in README).
Yes, the `joint-state-controller` package should be installed as well.

closed 2020-03-24 by vrontos

TracyLestrade, 2021-04-26:
yes,he is right. when u open new terminal. before run python script. do this. 
`source devel/setup.bash`