justagist/panda_simulator #5: Action client not connected: position_joint_trajectory_controller/follow_joint_trajectory

M-A-Robson, 2020-05-13:
timing out on the joint trajectory controller? see output below:

$ roslaunch panda_gazebo panda_world.launch 
... logging to /home/local/.ros/log/97cc4ace-9515-11ea-beed-3c6aa7cb24ab/roslaunch-PandaExpress-3151.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: in-order processing became default in ROS Melodic. You can drop the option.
started roslaunch server http://PandaExpress:45809/

SUMMARY
========

PARAMETERS
 * /SIMULATOR_: True
 * /arm/gravity_tip_name: panda_hand
 * /arm/root_name: panda_link0
 * /arm/tip_name: panda_hand
 * /controllers_config/default_controller: panda_simulator/p...
 * /controllers_config/position_controller: panda_simulator/p...
 * /controllers_config/torque_controller: panda_simulator/e...
 * /controllers_config/velocity_controller: panda_simulator/v...
 * /franka_gripper/robot_ip: sim
 * /gazebo/enable_ros_network: True
 * /gripper_config/default_grasp_epsilon/inner: 0.005
 * /gripper_config/default_grasp_epsilon/outer: 0.005
 * /gripper_config/default_speed: 0.1
 * /gripper_config/joint_names: ['panda_finger_jo...
 * /move_group/allow_trajectory_execution: True
 * /move_group/controller_list: [{'default': True...
 * /move_group/hand/planner_configs: ['SBLkConfigDefau...
 * /move_group/jiggle_fraction: 0.05
 * /move_group/max_range: 5.0
 * /move_group/max_safe_path_cost: 1
 * /move_group/moveit_controller_manager: moveit_simple_con...
 * /move_group/moveit_manage_controllers: True
 * /move_group/octomap_frame: camera_rgb_optica...
 * /move_group/octomap_resolution: 0.05
 * /move_group/panda_arm/planner_configs: ['SBLkConfigDefau...
 * /move_group/panda_arm_hand/planner_configs: ['SBLkConfigDefau...
 * /move_group/planner_configs/BFMTkConfigDefault/balanced: 0
 * /move_group/planner_configs/BFMTkConfigDefault/cache_cc: 1
 * /move_group/planner_configs/BFMTkConfigDefault/extended_fmt: 1
 * /move_group/planner_configs/BFMTkConfigDefault/heuristics: 1
 * /move_group/planner_configs/BFMTkConfigDefault/nearest_k: 1
 * /move_group/planner_configs/BFMTkConfigDefault/num_samples: 1000
 * /move_group/planner_configs/BFMTkConfigDefault/optimality: 1
 * /move_group/planner_configs/BFMTkConfigDefault/radius_multiplier: 1.0
 * /move_group/planner_configs/BFMTkConfigDefault/type: geometric::BFMT
 * /move_group/planner_configs/BKPIECEkConfigDefault/border_fraction: 0.9
 * /move_group/planner_configs/BKPIECEkConfigDefault/failed_expansion_score_factor: 0.5
 * /move_group/planner_configs/BKPIECEkConfigDefault/min_valid_path_fraction: 0.5
 * /move_group/planner_configs/BKPIECEkConfigDefault/range: 0.0
 * /move_group/planner_configs/BKPIECEkConfigDefault/type: geometric::BKPIECE
 * /move_group/planner_configs/BiESTkConfigDefault/range: 0.0
 * /move_group/planner_configs/BiESTkConfigDefault/type: geometric::BiEST
 * /move_group/planner_configs/BiTRRTkConfigDefault/cost_threshold: 1e300
 * /move_group/planner_configs/BiTRRTkConfigDefault/frountier_node_ratio: 0.1
 * /move_group/planner_configs/BiTRRTkConfigDefault/frountier_threshold: 0.0
 * /move_group/planner_configs/BiTRRTkConfigDefault/init_temperature: 100
 * /move_group/planner_configs/BiTRRTkConfigDefault/range: 0.0
 * /move_group/planner_configs/BiTRRTkConfigDefault/temp_change_factor: 0.1
 * /move_group/planner_configs/BiTRRTkConfigDefault/type: geometric::BiTRRT
 * /move_group/planner_configs/ESTkConfigDefault/goal_bias: 0.05
 * /move_group/planner_configs/ESTkConfigDefault/range: 0.0
 * /move_group/planner_configs/ESTkConfigDefault/type: geometric::EST
 * /move_group/planner_configs/FMTkConfigDefault/cache_cc: 1
 * /move_group/planner_configs/FMTkConfigDefault/extended_fmt: 1
 * /move_group/planner_configs/FMTkConfigDefault/heuristics: 0
[... 454 lines cut ...]
[topic_remap-10] killing on exit
[base_to_link0-5] killing on exit
[gazebo_gui-3] killing on exit
[robot_state_publisher-9] killing on exit
[gazebo-2] killing on exit
Traceback (most recent call last):
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/core.py", line 572, in signal_shutdown
    h()
  File "/home/local/ROS_packages/ws_moveit/src/panda_simulator/panda_sim_custom_action_server/scripts/start_joint_trajectory_server.py", line 61, in cleanup
    j.clean_shutdown()
  File "/home/local/ROS_packages/ws_moveit/src/panda_simulator/panda_sim_custom_action_server/src/panda_sim_custom_action_server/joint_trajectory_action_server.py", line 105, in clean_shutdown
    self._arm.exit_control_mode()
  File "/home/local/ROS_packages/ws_moveit/src/franka_ros_interface/franka_interface/src/franka_interface/arm.py", line 502, in exit_control_mode
    self.set_command_timeout(timeout)
  File "/home/local/ROS_packages/ws_moveit/src/franka_ros_interface/franka_interface/src/franka_interface/arm.py", line 537, in set_command_timeout
    self._pub_joint_cmd_timeout.publish(Float64(timeout))
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/topics.py", line 876, in publish
    raise ROSException("publish() to an unregistered() handle")
ROSException: publish() to an unregistered() handle
[gazebo_gui-3] escalating to SIGTERM
[gazebo-2] escalating to SIGTERM
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete

justagist, 2020-05-14:
The trajectory action server does not seem to be starting before the moveit request times out. Try running moveit_server seperately after the simulator is fully loaded.
```
$ roslaunch panda_gazebo panda_world.launch start_moveit:=false
$ roslaunch panda_sim_moveit sim_move_group.launch
```

M-A-Robson, 2020-05-15:
this works thanks

closed 2020-05-15 by M-A-Robson