justagist/panda_simulator #6: Python script for Action Server fails

henningdebuhr, 2020-06-17:
Hi,
i'm on U16.04 @ ROS kinetic
i followed the install instructions (i guess)
catkin_ws/src is:
CMakeLists.txt            
franka_ros_interface       
panda_robot
franka_panda_description  
orocos_kinematics_dynamics  
panda_simulator

during:
roslaunch panda_gazebo panda_world.launch
*lots of stuff*
then...
Starting context monitors...
[ INFO] [1592385325.219129687]: Starting scene monitor
[ INFO] [1592385325.225041161]: Listening to '/planning_scene'
[ INFO] [1592385325.225079605]: Starting world geometry monitor
publishing and latching message. Press ctrl-C to terminate
[ INFO] [1592385325.229153591]: Listening to '/collision_object' using message notifier with target frame '/world '
[ INFO] [1592385325.232247326]: Listening to '/planning_scene_world' for planning scene world geometry

[gripper_action_server_emulator-15] process has died [pid 19857, exit code -11, cmd /home/hdb/ai/catikin_ws/src/panda_simulator/panda_sim_custom_action_server/scripts/start_gripper_action_server.py __name:=gripper_action_server_emulator __log:=/home/hdb/.ros/log/12882122-b07b-11ea-9da0-185e0f9238df/gripper_action_server_emulator-15.log].
log file: /home/hdb/.ros/log/12882122-b07b-11ea-9da0-185e0f9238df/gripper_action_server_emulator-15*.log
[move_group-16] process has died [pid 19860, exit code -11, cmd /opt/ros/kinetic/lib/moveit_ros_move_group/move_group --debug __name:=move_group __log:=/home/hdb/.ros/log/12882122-b07b-11ea-9da0-185e0f9238df/move_group-16.log].
log file: /home/hdb/.ros/log/12882122-b07b-11ea-9da0-185e0f9238df/move_group-16*.log
[joint_trajectory_server_emulator-14] process has died [pid 19856, exit code -11, cmd /home/hdb/ai/catikin_ws/src/panda_simulator/panda_sim_custom_action_server/scripts/start_joint_trajectory_server.py __name:=joint_trajectory_server_emulator __log:=/home/hdb/.ros/log/12882122-b07b-11ea-9da0-185e0f9238df/joint_trajectory_server_emulator-14.log].
log file: /home/hdb/.ros/log/12882122-b07b-11ea-9da0-185e0f9238df/joint_trajectory_server_emulator-14*.log

this mentioned log files are empty
Any ideas?

justagist, 2020-06-17:
When the launch file runs, are the controllers starting correctly before the trajectory servers are loaded? You should see messages similar to `[controller_spawner_stopped-8] process has finished cleanly` and `[controller_spawner-7]  process has finished cleanly` in the [*lots of stuff*] part. 

If not, then the launch file is starting nodes in the wrong order. In this case, launch the simulator with the argument `use_custom_action_servers:=false`. Wait for the controllers to load fully, then start the server nodes (and if needed, the moveit server) in separate terminals (see launch files for the required nodes).

I can't be fully sure what is wrong without seeing an error log or fault trace of some kind, sorry.

henningdebuhr, 2020-06-17:
Manual start after  use_custom_action_servers:=false works. 

thx for support and upload

closed 2020-06-17 by justagist