justagist/panda_simulator #8: Moveit Execution

moritzknaust, 2020-07-13:
Hi @justagist!
I have a second question: The simulator should support MoveIt, right? I can plan trajectories like normal, but I cannot execute them on the simulator. 

Moveit wants to subscribe to a position_joint_trajectory_controller/follow_joint_trajectory but this fails, because the sim_controllers are used. But they do not support FollowJointTrajectory input. Is there a way to execute the trajectories on the simulator with one controller?  

I'm not sure if I just didn't find this option or if it is necessary to write a new FollowJointTrajectory controller using one of the sim-controllers.

justagist, 2020-07-13:
The simulator should start the moveit server by default. Are you able to run the `demo_moveit.launch` example?
Can you try the solution mentioned here: https://github.com/justagist/panda_simulator/issues/5#issuecomment-628505799

moritzknaust, 2020-07-14:
I tried the solution in #5 and now I can run the demo!

Here is exactly what I have done:
- comment out the `<include ... panda.world>` block in `demo_moveit.launch`
- launch all parts separately:
```
roslaunch panda_gazebo panda_world.launch start_moveit:=false
roslaunch panda_sim_moveit sim_move_group.launch
roslaunch panda_simulator_examples demo_moveit.launch
```
It seems to be a timing problem, but nothing really important.

And now I also understand how it works. It is a very nice solution to use one action server for all controller types!

closed 2020-07-14 by moritzknaust