justagist/panda_simulator #12: [gripper_action_server_emulator-12] process has died

Sission, 2020-11-03:
Hey, I tried to run:
$ roslaunch panda_gazebo panda_world.launch start_moveit:=false

It can generate the robot in the gazebo successfully, however, with these errors:
[gripper_action_server_emulator-12] process has died [pid 12790, exit code 1, cmd /home/sission/workspace/src/panda_simulator/panda_sim_custom_action_server/scripts/start_gripper_action_server.py __name:=gripper_action_server_emulator __log:=/home/sission/.ros/log/5c634612-1e05-11eb-801a-049226c26561/gripper_action_server_emulator-12.log].
log file: /home/sission/.ros/log/5c634612-1e05-11eb-801a-049226c26561/gripper_action_server_emulator-12*.log

[joint_trajectory_server_emulator-11] process has died [pid 12782, exit code 1, cmd /home/sission/workspace/src/panda_simulator/panda_sim_custom_action_server/scripts/start_joint_trajectory_server.py __name:=joint_trajectory_server_emulator __log:=/home/sission/.ros/log/5c634612-1e05-11eb-801a-049226c26561/joint_trajectory_server_emulator-11.log].
log file: /home/sission/.ros/log/5c634612-1e05-11eb-801a-049226c26561/joint_trajectory_server_emulator-11*.log

Also, the generated robot seems to have uncorrect gravity (not the pose you shown in the gif)

Sission, 2020-11-03:
[ WARN] [1604431878.750975787, 5.182000000]: Waiting for position_joint_trajectory_controller/follow_joint_trajectory to come up
[ WARN] [1604431884.756394904, 11.182000000]: Waiting for position_joint_trajectory_controller/follow_joint_trajectory to come up
[ERROR] [1604431890.760980570, 17.182000000]: Action client not connected: position_joint_trajectory_controller/follow_joint_trajectory
[ WARN] [1604431895.780569657, 22.197000000]: Waiting for franka_gripper/gripper_action to come up
[ WARN] [1604431901.785184544, 28.197000000]: Waiting for franka_gripper/gripper_action to come up
[ERROR] [1604431907.789568332, 34.197000000]: Action client not connected: franka_gripper/gripper_action
[ INFO] [1604431907.886445766, 34.294000000]: Returned 0 controllers in list
[ INFO] [1604431907.891737918, 34.299000000]: Trajectory execution is managing controllers
Loading 'move_group/ApplyPlanningSceneService'...
Loading 'move_group/ClearOctomapService'...
Loading 'move_group/MoveGroupCartesianPathService'...
Loading 'move_group/MoveGroupExecuteTrajectoryAction'...
Loading 'move_group/MoveGroupGetPlanningSceneService'...
Loading 'move_group/MoveGroupKinematicsService'...
Loading 'move_group/MoveGroupMoveAction'...
Loading 'move_group/MoveGroupPickPlaceAction'...
Loading 'move_group/MoveGroupPlanService'...
Loading 'move_group/MoveGroupQueryPlannersService'...
Loading 'move_group/MoveGroupStateValidationService'...
[ INFO] [1604431907.920488882, 34.328000000]: 

********************************************************
* MoveGroup using: 
*     - ApplyPlanningSceneService
*     - ClearOctomapService
*     - CartesianPathService
*     - ExecuteTrajectoryAction
*     - GetPlanningSceneService
*     - KinematicsService
*     - MoveAction
*     - PickPlaceAction
*     - MotionPlanService
*     - QueryPlannersService
*     - StateValidationService
********************************************************

[ INFO] [1604431907.920518562, 34.328000000]: MoveGroup context using planning plugin ompl_interface/OMPLPlanner
[ INFO] [1604431907.920534836, 34.328000000]: MoveGroup context initialization complete

justagist, 2020-11-03:
Hi, 
Please make sure you are using the right command: `roslaunch panda_gazebo panda_world.launch start_moveit:=false`. From the error message you shared, it looks like moveit server is still starting which it shouldn't.
If you are doing this correctly and you still have issue, please share the full error log. Also include the log files if available. 

The pose shown in the gif may not necessarily match how it starts in your case; it may be different depending on how gazebo reads the urdf. This is not because of gravity necessarily. Also the gravity in the simulator is disabled by default.

Sission, 2020-11-03:
Yes, I do use the right command, there is just one error now. Here is the details:

sission@Ros:~/workspace$ roslaunch panda_gazebo panda_world.launch start_moveit:=false
... logging to /home/sission/.ros/log/fd8643a6-1e1d-11eb-b53e-049226c26561/roslaunch-Ros-11227.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: in-order processing became default in ROS Melodic. You can drop the option.
started roslaunch server http://Ros:44451/

SUMMARY
========

PARAMETERS
 * /SIMULATOR_: True
 * /arm/gravity_tip_name: panda_hand
 * /arm/root_name: panda_link0
 * /arm/tip_name: panda_hand
 * /controllers_config/default_controller: panda_simulator/p...
 * /controllers_config/position_controller: panda_simulator/p...
 * /controllers_config/torque_controller: panda_simulator/e...
 * /controllers_config/velocity_controller: panda_simulator/v...
 * /franka_gripper/robot_ip: sim
 * /gazebo/enable_ros_network: True
 * /gripper_config/default_grasp_epsilon/inner: 0.005
 * /gripper_config/default_grasp_epsilon/outer: 0.005
 * /gripper_config/default_speed: 0.1
 * /gripper_config/joint_names: ['panda_finger_jo...
 * /panda_simulator/effort_joint_gravity_controller/joints/panda_joint1_controller/joint: panda_joint1
 * /panda_simulator/effort_joint_gravity_controller/joints/panda_joint2_controller/joint: panda_joint2
 * /panda_simulator/effort_joint_gravity_controller/joints/panda_joint3_controller/joint: panda_joint3
 * /panda_simulator/effort_joint_gravity_controller/joints/panda_joint4_controller/joint: panda_joint4
 * /panda_simulator/effort_joint_gravity_controller/joints/panda_joint5_controller/joint: panda_joint5
 * /panda_simulator/effort_joint_gravity_controller/joints/panda_joint6_controller/joint: panda_joint6
 * /panda_simulator/effort_joint_gravity_controller/joints/panda_joint7_controller/joint: panda_joint7
 * /panda_simulator/effort_joint_gravity_controller/type: panda_sim_control...
 * /panda_simulator/effort_joint_torque_controller/joints/panda_joint1_controller/joint: panda_joint1
 * /panda_simulator/effort_joint_torque_controller/joints/panda_joint2_controller/joint: panda_joint2
 * /panda_simulator/effort_joint_torque_controller/joints/panda_joint3_controller/joint: panda_joint3
 * /panda_simulator/effort_joint_torque_controller/joints/panda_joint4_controller/joint: panda_joint4
 * /panda_simulator/effort_joint_torque_controller/joints/panda_joint5_controller/joint: panda_joint5
 * /panda_simulator/effort_joint_torque_controller/joints/panda_joint6_controller/joint: panda_joint6
 * /panda_simulator/effort_joint_torque_controller/joints/panda_joint7_controller/joint: panda_joint7
 * /panda_simulator/effort_joint_torque_controller/topic: /panda_simulator/...
 * /panda_simulator/effort_joint_torque_controller/type: panda_sim_control...
 * /panda_simulator/joint_state_controller/publish_rate: 100
 * /panda_simulator/joint_state_controller/type: joint_state_contr...
 * /panda_simulator/panda_gripper_controller/joints/panda_finger_joint1_controller/joint: panda_finger_joint1
 * /panda_simulator/panda_gripper_controller/joints/panda_finger_joint1_controller/pid/d: 0.5
 * /panda_simulator/panda_gripper_controller/joints/panda_finger_joint1_controller/pid/i: 50
 * /panda_simulator/panda_gripper_controller/joints/panda_finger_joint1_controller/pid/p: 5000
 * /panda_simulator/panda_gripper_controller/joints/panda_finger_joint2_controller/joint: panda_finger_joint2
 * /panda_simulator/panda_gripper_controller/joints/panda_finger_joint2_controller/pid/d: 0.5
 * /panda_simulator/panda_gripper_controller/joints/panda_finger_joint2_controller/pid/i: 50
 * /panda_simulator/panda_gripper_controller/joints/panda_finger_joint2_controller/pid/p: 5000
 * /panda_simulator/panda_gripper_controller/pub_topic: /franka_gripper/j...
 * /panda_simulator/panda_gripper_controller/type: panda_sim_control...
 * /panda_simulator/position_joint_position_controller/joints/panda_joint1_controller/joint: panda_joint1
 * /panda_simulator/position_joint_position_controller/joints/panda_joint1_controller/pid/d: 25.0
 * /panda_simulator/position_joint_position_controller/joints/panda_joint1_controller/pid/i: 10.0
 * /panda_simulator/position_joint_position_controller/joints/panda_joint1_controller/pid/p: 3000
 * /panda_simulator/position_joint_position_controller/joints/panda_joint2_controller/joint: panda_joint2
 * /panda_simulator/position_joint_position_controller/joints/panda_joint2_controller/pid/d: 30.0
 * /panda_simulator/position_joint_position_controller/joints/panda_joint2_controller/pid/i: 10.0
 * /panda_simulator/position_joint_position_controller/joints/panda_joint2_controller/pid/p: 3000
 * /panda_simulator/position_joint_position_controller/joints/panda_joint3_controller/joint: panda_joint3
 * /panda_simulator/position_joint_position_controller/joints/panda_joint3_controller/pid/d: 30.0
 * /panda_simulator/position_joint_position_controller/joints/panda_joint3_controller/pid/i: 10.0
 * /panda_simulator/position_joint_position_controller/joints/panda_joint3_controller/pid/p: 3000
[... 203 lines cut ...]
[ INFO] [1604439984.483557791, 0.371000000]: Loading sub-controller 'panda_joint4_controller', Namespace: /panda_simulator/effort_joint_gravity_controller/joints/panda_joint4_controller
[... cut ...]
log file: /home/sission/.ros/log/fd8643a6-1e1d-11eb-b53e-049226c26561/panda_gripper_controller_spawner_stopped-9*.log
[controller_spawner-7] process has finished cleanly
log file: /home/sission/.ros/log/fd8643a6-1e1d-11eb-b53e-049226c26561/controller_spawner-7*.log
[controller_spawner_stopped-8] process has finished cleanly
log file: /home/sission/.ros/log/fd8643a6-1e1d-11eb-b53e-049226c26561/controller_spawner_stopped-8*.log
[robot_description-4] process has finished cleanly
log file: /home/sission/.ros/log/fd8643a6-1e1d-11eb-b53e-049226c26561/robot_description-4*.log

Sission, 2020-11-03:
Here is the log file by using: roslaunch panda_gazebo panda_world.launch

sission@Ros:~/workspace/src$ roslaunch panda_gazebo panda_world.launch
... logging to /home/sission/.ros/log/d9d006f8-1e1e-11eb-b53e-049226c26561/roslaunch-Ros-12369.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: in-order processing became default in ROS Melodic. You can drop the option.
started roslaunch server http://Ros:41215/

SUMMARY
========

PARAMETERS
 * /SIMULATOR_: True
 * /arm/gravity_tip_name: panda_hand
 * /arm/root_name: panda_link0
 * /arm/tip_name: panda_hand
 * /controllers_config/default_controller: panda_simulator/p...
 * /controllers_config/position_controller: panda_simulator/p...
 * /controllers_config/torque_controller: panda_simulator/e...
 * /controllers_config/velocity_controller: panda_simulator/v...
 * /franka_gripper/robot_ip: sim
 * /gazebo/enable_ros_network: True
 * /gripper_config/default_grasp_epsilon/inner: 0.005
 * /gripper_config/default_grasp_epsilon/outer: 0.005
 * /gripper_config/default_speed: 0.1
 * /gripper_config/joint_names: ['panda_finger_jo...
 * /move_group/allow_trajectory_execution: True
 * /move_group/controller_list: [{'default': True...
 * /move_group/hand/planner_configs: ['SBLkConfigDefau...
 * /move_group/jiggle_fraction: 0.05
 * /move_group/max_range: 5.0
 * /move_group/max_safe_path_cost: 1
 * /move_group/moveit_controller_manager: moveit_simple_con...
 * /move_group/moveit_manage_controllers: True
 * /move_group/octomap_frame: camera_rgb_optica...
 * /move_group/octomap_resolution: 0.05
 * /move_group/panda_arm/planner_configs: ['SBLkConfigDefau...
 * /move_group/panda_arm_hand/planner_configs: ['SBLkConfigDefau...
 * /move_group/planner_configs/BFMTkConfigDefault/balanced: 0
 * /move_group/planner_configs/BFMTkConfigDefault/cache_cc: 1
 * /move_group/planner_configs/BFMTkConfigDefault/extended_fmt: 1
 * /move_group/planner_configs/BFMTkConfigDefault/heuristics: 1
 * /move_group/planner_configs/BFMTkConfigDefault/nearest_k: 1
 * /move_group/planner_configs/BFMTkConfigDefault/num_samples: 1000
 * /move_group/planner_configs/BFMTkConfigDefault/optimality: 1
 * /move_group/planner_configs/BFMTkConfigDefault/radius_multiplier: 1.0
 * /move_group/planner_configs/BFMTkConfigDefault/type: geometric::BFMT
 * /move_group/planner_configs/BKPIECEkConfigDefault/border_fraction: 0.9
 * /move_group/planner_configs/BKPIECEkConfigDefault/failed_expansion_score_factor: 0.5
 * /move_group/planner_configs/BKPIECEkConfigDefault/min_valid_path_fraction: 0.5
 * /move_group/planner_configs/BKPIECEkConfigDefault/range: 0.0
 * /move_group/planner_configs/BKPIECEkConfigDefault/type: geometric::BKPIECE
 * /move_group/planner_configs/BiESTkConfigDefault/range: 0.0
 * /move_group/planner_configs/BiESTkConfigDefault/type: geometric::BiEST
 * /move_group/planner_configs/BiTRRTkConfigDefault/cost_threshold: 1e300
 * /move_group/planner_configs/BiTRRTkConfigDefault/frountier_node_ratio: 0.1
 * /move_group/planner_configs/BiTRRTkConfigDefault/frountier_threshold: 0.0
 * /move_group/planner_configs/BiTRRTkConfigDefault/init_temperature: 100
 * /move_group/planner_configs/BiTRRTkConfigDefault/range: 0.0
 * /move_group/planner_configs/BiTRRTkConfigDefault/temp_change_factor: 0.1
 * /move_group/planner_configs/BiTRRTkConfigDefault/type: geometric::BiTRRT
 * /move_group/planner_configs/ESTkConfigDefault/goal_bias: 0.05
 * /move_group/planner_configs/ESTkConfigDefault/range: 0.0
 * /move_group/planner_configs/ESTkConfigDefault/type: geometric::EST
 * /move_group/planner_configs/FMTkConfigDefault/cache_cc: 1
 * /move_group/planner_configs/FMTkConfigDefault/extended_fmt: 1
 * /move_group/planner_configs/FMTkConfigDefault/heuristics: 0
[... 433 lines cut ...]
Loading 'move_group/MoveGroupPickPlaceAction'...
Loading 'move_group/MoveGroupPlanService'...
Loading 'move_group/MoveGroupQueryPlannersService'...
Loading 'move_group/MoveGroupStateValidationService'...
[ INFO] [1604440379.036871833, 34.507000000]: 

********************************************************
* MoveGroup using: 
*     - ApplyPlanningSceneService
*     - ClearOctomapService
*     - CartesianPathService
*     - ExecuteTrajectoryAction
*     - GetPlanningSceneService
*     - KinematicsService
*     - MoveAction
*     - PickPlaceAction
*     - MotionPlanService
*     - QueryPlannersService
*     - StateValidationService
********************************************************

[ INFO] [1604440379.036905074, 34.507000000]: MoveGroup context using planning plugin ompl_interface/OMPLPlanner
[ INFO] [1604440379.036916532, 34.507000000]: MoveGroup context initialization complete

You can start planning now!

justagist, 2020-11-03:
As it says in the error message, quaternion module is not installed. This could be installed by following the instructions in the readme. You can run `pip install numpy-quaternion` to fix this. Everything else looks fine.
If you want to use moveit server with the simulator, follow the instructions given in the readme's "Known Issues" section.

Sission, 2020-11-03:
Thank you so much!
I know it is not this repo's issue, but can I ask this question here? I am sorry I am not very good at Ros and Linux.

I tried 
sission@Ros:~/workspace/devel$ pip install numpy-quaternion
Defaulting to user installation because normal site-packages is not writeable
Requirement already satisfied: numpy-quaternion in /home/sission/.local/lib/python3.6/site-packages (2020.11.2.17.0.49)
Requirement already satisfied: numpy>=1.13 in /usr/local/lib/python3.6/dist-packages (from numpy-quaternion) (1.19.2)

And I checked my $PYTHONPATH
sission@Ros:~/workspace/devel$ echo $PYTHONPATH
/home/sission/workspace/devel/lib/python2.7/dist-packages:/home/sission/workspace/devel/lib/python2.7/dist-packages:/opt/ros/melodic/lib/python2.7/dist-packages

Could you help me how to make that install in my current pythonpath so the program can run? I failed after googling it. Thank you!

justagist, 2020-11-03:
1. Run the install command with `sudo` (`sudo pip install numpy-quaternion`) or install it for your user. Check if this works. 
2. If it still installs for python 3, make sure you have installed pip for python 2 (`sudo apt install python-pip`). Then try step 1 again. 
3. If there is still no difference, force use python 2 pip (`sudo pip2 install... `).

Sission, 2020-11-03:
THANK YOU SOOOOOOOOOOOOO MUCH! YOU ARE MY LIFE SAVER!!!!

closed 2020-11-04 by justagist