justagist/panda_simulator #15: Unable to identify any set of controllers

Sission, 2020-11-08:
Hi! I tried to run the  demo for testing the moveit planner interface with the simulated robot
roslaunch panda_simulator_examples demo_moveit.launch
However, it shows the following error.

[ERROR] [1604853219.157810719, 59.716000000]: Unable to identify any set of controllers that can actuate the specified joints: [ panda_joint1 panda_joint2 panda_joint3 panda_joint4 panda_joint5 panda_joint6 panda_joint7 ]
[ERROR] [1604853219.157828995, 59.716000000]: Known controllers and their joints:
controller 'franka_gripper' controls joints:
  panda_finger_joint1
  panda_finger_joint2

[ERROR] [1604853219.157858878, 59.716000000]: Apparently trajectory initialization failed

Another demo roslaunch panda_simulator_examples demo_task_space_control.launch
works fine.

justagist, 2020-11-09:
I'm not sure what is happening. It looks like the trajectory action server is not running for the arm. Can you tell me the commands you ran before the demo command?
Are you running:
```bash
$ roslaunch panda_gazebo panda_world.launch start_moveit:=false
$ roslaunch panda_sim_moveit sim_move_group.launch
$ roslaunch panda_simulator_examples demo_moveit.launch
```

Sission, 2020-11-10:
I did. I believe it is my Ubuntu system problem. Sometimes it works, sometimes it doesn't...

closed 2020-11-10 by Sission

justagist, 2020-11-10:
I see... Maybe the action client fails to start in time in your system.
Try starting all tasks separately:
```bash
$ roslaunch panda_gazebo panda_world.launch use_custom_action_servers:=false
$ rosrun panda_sim_custom_action_server start_joint_trajectory_server.py # wait for message: "Joint Trajectory Action Server Running."
$ rosrun panda_sim_custom_action_server start_gripper_action_server.py # wait for message: "GripperInterface: Gripper action servers found!"
$ roslaunch panda_sim_moveit sim_move_group.launch
$ roslaunch panda_simulator_examples demo_moveit.launch
```

Caramel11, 2024-08-27:
> 我明白了...也许您的系统中的操作未能及时启动。 请尝试分别启动所有任务：
> 
> ```shell
> $ roslaunch panda_gazebo panda_world.launch use_custom_action_servers:=false
> $ rosrun panda_sim_custom_action_server start_joint_trajectory_server.py # wait for message: "Joint Trajectory Action Server Running."
> $ rosrun panda_sim_custom_action_server start_gripper_action_server.py # wait for message: "GripperInterface: Gripper action servers found!"
> $ roslaunch panda_sim_moveit sim_move_group.launch
> $ roslaunch panda_simulator_examples demo_moveit.launch
> ```
> 
> 如果此方法有效，则意味着您在报表服务器上线之前启动了 moveit 服务器。因此，您只需运行我在之前的评论中分享的 3 个命令，但请等到看到提示“动作正在服务器运行”的消息恢复后启动 moveit 服务器。

It also solved my problem! thank you!