justagist/panda_simulator #16: Build fails on melodic

adi243, 2020-11-13:
Hi,

Great work on the simulator, I was trying to build the package and saw the following error. There seems to be a library file missing. Not really sure what I am doing wrong. Any help would be great.

/catkin_ws/src/franka_ros_interface/franka_interface/src/franka_control_node.cpp:44:10: fatal error: franka_control/ErrorRecoveryAction.h: No such file or directory
 #include <franka_control/ErrorRecoveryAction.h>
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/custom_franka_control_node.dir/src/franka_control_node.cpp.o] Error 1
make[1]: *** [CMakeFiles/custom_franka_control_node.dir/all] Error 2
make: *** [all] Error 2

justagist, 2020-11-13:
Hi,
This is because the official franka-ros package version was recently upgraded, and the franka_ros_interface package does not officially support the new version yet. There are two options here:
1. (not fully tested) There is a development branch (v0.7.1) for the franka_ros_interface repository which is targeted for the new version of franka_ros. You can switch to this branch and try a clean build of the workspace again.
2. (tested) You will have to build the older version of franka_ros from source. To do this, clone the [franka_ros](https://github.com/frankaemika/franka_ros), checkout to the previous stable version (`git checkout 49e5ac1`), and rebuild workspace. 
Hope this helped. I will update the Readme soon

adi243, 2020-11-13:
Builds fine now. Thank you

closed 2020-11-13 by adi243