justagist/panda_simulator #19: Launch crashes with protobuf error.

padmaja-kulkarni, 2020-12-01:
Hi There,

Thank you for this code and the repo. 

I was able to build this repo. But when I try to launch `roslaunch panda_gazebo panda_world.launch`, I get the following error and the launch crashes. I am using another robots with Gazebo and it works fine. I am using ROS melodic with Gazebo 9. By Default, Gazebo needs protoc 3.0.0, which is also system installed. If I install 3.14 then the program does not compile. Any insights on this issue will be very helpful. 

Thanks a ton! 

 `[ INFO] [1606816804.851503670, 0.020000000]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[ INFO] [1606816804.867844528, 0.035000000]: Physics dynamic reconfigure ready.
[ INFO] [1606816805.088884166, 0.153000000]: Loading gazebo_ros_control plugin
[ INFO] [1606816805.091071827, 0.153000000]: Starting gazebo_ros_control plugin in namespace: 
[ INFO] [1606816805.091505893, 0.153000000]: gazebo_ros_control plugin is waiting for model URDF in parameter [/robot_description] on the ROS param server.
[libprotobuf FATAL google/protobuf/stubs/common.cc:67] This program requires version 3.14.0 of the Protocol Buffer runtime library, but the installed version is 3.0.0.  Please update your library.  If you compiled the program yourself, make sure that your headers are from the same version of Protocol Buffers as your link-time library.  (Version verification failed in "google/protobuf/any.pb.cc".)
terminate called after throwing an instance of 'google::protobuf::FatalException'
  what():  This program requires version 3.14.0 of the Protocol Buffer runtime library, but the installed version is 3.0.0.  Please update your library.  If you compiled the program yourself, make sure that your headers are from the same version of Protocol Buffers as your link-time library.  (Version verification failed in "google/protobuf/any.pb.cc".)
Aborted (core dumped)
[gazebo-1] process has died [pid 31958, exit code 134, cmd /opt/ros/melodic/lib/gazebo_ros/gzserver -e ode /home/padmaja/ros/src/panda_simulator/panda_gazebo/worlds/panda.world __name:=gazebo __log:=/home/padmaja/.ros/log/a9a6cd74-33bb-11eb-8c49-50e085401a14/gazebo-1.log].
log file: /home/padmaja/.ros/log/a9a6cd74-33bb-11eb-8c49-50e085401a14/gazebo-1*.log
[ERROR] [1606816806.069719, 0.153000]: SetModelConfiguration service failed. Exiting.
[robot_description-3] process has died [pid 31966, exit code 1, cmd /opt/ros/melodic/lib/gazebo_ros/spawn_model -param robot_description -urdf -model panda -J panda::panda_joint1 0.000 -J panda::panda_joint2 -0.785 -J panda::panda_joint3 0.0 -J panda::panda_joint4 -2.356 -J panda::panda_joint5 0.0 -J panda::panda_joint6 1.57 -J panda::panda_joint7 0.785 __name:=robot_description __log:=/home/padmaja/.ros/log/a9a6cd74-33bb-11eb-8c49-50e085401a14/robot_description-3.log].
log file: /home/padmaja/.ros/log/a9a6cd74-33bb-11eb-8c49-50e085401a14/robot_description-3*.log
`

justagist, 2020-12-01:
Hi,
It looks like there may be conflicting versions of protobuf installed. You might want to remove the older versions manually. I am not sure what is happening as I cannot replicate the error. Another issue could be the compatibility of the version of ignition installed. Apparently, there are a lot of compatibility issues between versions of gazebo, ignition and protobuf ([ref](https://github.com/uuvsimulator/uuv_simulator/issues/287)).
Sorry for not being of much help.

padmaja-kulkarni, 2020-12-01:
Thank you so much for your quick reply. After a long process of deleting the separate version, I am finally moved onto a different error: :/

`[ERROR] [1606823553.276448, 0.744000]: SetModelConfiguration service failed. Exiting.`

I will close the issue from here. :)

closed 2020-12-01 by padmaja-kulkarni

justagist, 2020-12-01:
I can't really say much from looking at the error message. Maybe if you share the error trace or full log, I can take a look, but I can't say I have seen this before.
Hope you can fix the issue!

padmaja-kulkarni, 2020-12-03:
I could close the issue. This one was an easy fix. Thank you very much for your quick replies.

yhindka, 2021-11-17:
Do you remember how you fixed the SetModelConfiguration service failed error? I'm getting the same error and can't figure out how to fix it.

apandastider, 2021-12-17:
Can anyone please guide me how did you the error : SetModelConfiguration service failed. Exiting. ? It will be very helpful for me.