justagist/panda_simulator #21: Can't pick up the objects in Gazebo

pytalha, 2020-12-23:
I want to simulate basic pick and place using Panda arm in Gazebo. I can move the arm very well in Gazebo using Rviz but I don't have absolute control over gripper in Rviz except open and close. 

However I've spawned the arm in my world and I'm able to execute Rviz motion plan in Gazebo but the gripper can't grip the object. The attached video demonstrates the problem well. 

Please guide how to fix this problem and if I want to control the gripper manually using ROS and embed it in my motion plan, how can that be done? Would really appreciate the help :) 

https://user-images.githubusercontent.com/48376895/102995851-e1a61d80-455c-11eb-83af-b40d35dfa9b1.mp4

justagist, 2020-12-28:
Hi, It looks like the object you are trying to interact with does not have collision enabled/defined. If you are loading it from a urdf, make sure that the collision tag is defined. This is unlikely to be an issue of this package. I am attaching a video of the grasping in action to show you how it should look with a defined object.

https://user-images.githubusercontent.com/22923089/103212090-d6a71080-4901-11eb-9e2a-13c134c4dbe0.mp4

closed 2020-12-29 by justagist

pytalha, 2020-12-30:
I believe the collision tags are defined in both box_1 and box_2. Can you please point out the mistake in the world file?

<?xml version="1.0" ?>
<sdf version="1.6">
  <world name="default">
    <include>
      <uri>model://ground_plane</uri>
    </include>

    <include>
      <uri>model://sun</uri>
    </include>

    <physics type="ode">
      <real_time_update_rate>1000.0</real_time_update_rate>
    </physics>
    
    <gravity>
      0.0 0.0 0.0
    </gravity>

    <gui fullscreen='0'>
    </gui>

    
    <model name='cardboard_box1'>
      <static>1</static>
      <link name='link'>
        <pose frame=''>0.5 0.6 0 0 0 0</pose>
        <visual name='visual'>
          <geometry>
            <mesh>
              <uri>model://cardboard_box/meshes/cardboard_box.dae</uri>
              <scale>1 1 1</scale>
            </mesh>
          </geometry>
        </visual>
      </link>
    </model>

    
    <model name='cardboard_box2'>
      <static>1</static>
      <link name='link'>
        <pose frame=''>-0.5 -0.6 0 0 0 0</pose>
        <visual name='visual'>
          <geometry>
            <mesh>
              <uri>model://cardboard_box/meshes/cardboard_box.dae</uri>
              <scale>1 1 1</scale>
            </mesh>
          </geometry>
        </visual>
      </link>
    </model>

    
    <model name='box_1'>
      <static>1</static>
      <link name='link'>
        <pose frame=''>0.6 0.5 0.4 0 0 0</pose>
        <visual name='visual'>
          <geometry>
            <mesh>
              <uri>model://cardboard_box/meshes/cardboard_box.dae</uri>
              <scale>0.1 0.1 0.1</scale>
            </mesh>
          </geometry>
        </visual>
        <collision name='block'>
          <pose frame=''>0 0 0 0 0 0</pose>
        <geometry> 
          <box>
            <size>0.05 0.05 0.05</size>
          </box>
        </geometry>
        <max_contacts>10</max_contacts>
      </collision>
      <self_collide>0</self_collide>
      <kinematic>0</kinematic>
      </link>
    </model>

    <model name='box_2'>
      <static>1</static>
      <link name='link'>
        <pose frame=''>0.4 0.5 0.4 0 0 0</pose>
        <visual name='visual'>
          <geometry>
            <mesh>
              <uri>model://cardboard_box/meshes/cardboard_box.dae</uri>
              <scale>0.1 0.1 0.1</scale>
            </mesh>
          </geometry>
        </visual>
        <collision name='block'>
          <pose frame=''>0 0 0 0 0 0</pose>
        <geometry> 
          <box>
            <size>0.05 0.05 0.05</size>
          </box>
        </geometry>
        <max_contacts>10</max_contacts>
      </collision>
      <self_collide>0</self_collide>
      <kinematic>0</kinematic>
      </link>
    </model>

  </world>
</sdf>

kapil13j, 2023-02-22:
Did you find out the exact solution to this problem?