justagist/panda_simulator #25: `panda_gazebo_ros_control_plugin` does not provide a working `EffortJointInterface`

smihael, 2021-02-18:
When I launch the simulator using [panda_world.launch](https://github.com/justagist/panda_simulator/blob/melodic-devel/panda_gazebo/launch/panda_world.launch) `panda_gazebo_ros_control_plugin` gets loaded instead of default gazebo `ros_control` plugin.

It seems to provide `EffortJointInterface` for ros_control plugins. Correct me if I'm wrong.

Let's say I have a plugin which uses this interface
`class TestController : public  controller_interface::MultiInterfaceController<hardware_interface::EffortJointInterface>`.
As usual I get the joint handles
`joint_handles[i] = effort_joint_interface->getHandle(joint_names[i]))`.  

They can be used for getting the for example the joint position with `joint_handles_[i].getPosition()`, but when I try to set effort with `joint_handles_[i].setCommand(tau[i])` the robot does not move.

The same code works on the real robot and in [an older revision of a similar project](https://github.com/erdalpekel/panda_simulation/tree/cde500f9ce233a5c0333798e1a8cdef370bcd97e).

For this reason I'm unable to run the simulator with a variety of other ros controllers.

If you load the following yaml to the paramter server:

```
panda_arm_controller:
    type: effort_controllers/JointTrajectoryController
    joints:
        - panda_joint1
        - panda_joint2
        - panda_joint3
        - panda_joint4
        - panda_joint5
        - panda_joint6
        - panda_joint7

    gains:
        panda_joint1: { p: 12000, d: 50, i: 0.0, i_clamp: 10000 }
        panda_joint2: { p: 30000, d: 100, i: 0.02, i_clamp: 10000 }
        panda_joint3: { p: 18000, d: 50, i: 0.01, i_clamp: 1 }
        panda_joint4: { p: 18000, d: 70, i: 0.01, i_clamp: 10000 }
        panda_joint5: { p: 12000, d: 70, i: 0.01, i_clamp: 1 }
        panda_joint6: { p: 7000, d: 50, i: 0.01, i_clamp: 1 }
        panda_joint7: { p: 2000, d: 20, i: 0.0, i_clamp: 1 }

    constraints:
        goal_time: 2.0

    state_publish_rate: 25
```

And start `panda_arm_controller` with: `rosrun controller_manager spawner panda_arm_controller` 
you can use `rqt-joint-trajectory-controller` GUI to move the robot in erdalpekel's `panda_simulation`, but not in this project.

justagist, 2021-02-19:
This package has a custom control plugin that it uses instead of the default gazebo_ros_control. The plugin uses the custom PandaRobotHWSim interface that is defined for this package instead of DefaultRobotHWSim (gazebo's native hw interface, which provides the `hardware_interface::EffortJointInterface` that you seem to need). If you want to write controllers for this simulator directly, you will have to use `panda_sim_controllers::JointArrayController<panda_effort_controllers::JointEffortController>` instead (see the defined controllers in this project). This is not really recommended if you want to write your own low-level gazebo controllers because the custom hw interface will make it unnecessarily complicated. You can use the default gazebo controllers as you mentioned in #26. 
This package was not intended for providing control interfaces to write low-level controllers, but for providing a simulated platform for emulating the robot when using the [FrankaROSInterface](https://github.com/justagist/franka_ros_interface) API. This meant defining several custom interfaces and features to make the simulation respond to different API calls (in the form of topics/services), as well as having custom controller management setting that can automatically switch controllers depending on the API calls made by the user, and hence the need for custom HW interface and controller plugin. This allows to control both the real and simulated robot using all of the defined low-level controllers (position, velocity, torque) using the same python API or through ROS topics/services.

> For this reason I'm unable to run the simulator with a variety of other ros controllers.

It is possible to define custom controllers in python over the low-level controllers that are defined. But if you wish to define your own low-level controllers directly, then this package may not be suitable for you.

> And start `panda_arm_controller` with: `rosrun controller_manager spawner panda_arm_controller`
you can use `rqt-joint-trajectory-controller` GUI to move the robot in erdalpekel's `panda_simulation`, but not in this project.

Again, since this package uses a custom control plugin, it is not possible to use the default ros controllers (such as effort_controllers/JointTrajectoryController) together with the custom controllers defined in the plugin. However, I have defined a trajectory server to emulate the behaviour of this controller which is needed for MoveIt, and this works as required when using the API.

If you wish to define custom controllers and bypass the utilities provided by the franka_ros_interface package, then you can use the default interface provided by gazebo, which is easier to use. In this case, you can use erdalpekel's `panda_simulation` package or [gazebo_panda](https://github.com/justagist/gazebo_panda), which uses the same robot descriptions as this simulator but uses the default gazebo_ros controls without the overhead of or dependency on franka_ros_interface.

closed 2021-02-19 by justagist

smihael, 2021-04-06:
Thank you for pointing to [gazebo_panda](https://github.com/justagist/gazebo_panda).

For anyone interested, I prepared an initial version of a custom Gazebo plugin that retains default behavior of the `hardware_interface::EffortJointInterface` (among others) and adds an (incomplete) emulation of the `franka_model_interface` & `franka_state_interface` at: https://github.com/smihael/panda_sim_hw

In combination with `gazebo_panda` it is now possible to run [example controllers](https://github.com/frankaemika/franka_ros/tree/kinetic-devel/franka_example_controllers) from franka_ros in simulation.

justagist, 2021-04-06:
That is fantastic! I'll definitely try it out at some point soon. Thanks for sharing!