justagist/panda_simulator #27: Fail to startup

rouzinho, 2021-02-22:
Hi !
NIce work there but I can't startup the simulation. Or actually gazebo start but end up crashing before starting moveit...

```
... logging to /home/altair/.ros/log/c59f8780-753f-11eb-8ab4-3c6aa7b729ab/roslaunch-altair-88694.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: in-order processing became default in ROS Melodic. You can drop the option.
started roslaunch server http://altair:35151/

SUMMARY
========

PARAMETERS
 * /SIMULATOR_: True
 * /arm/gravity_tip_name: panda_hand
 * /arm/root_name: panda_link0
 * /arm/tip_name: panda_hand
 * /controllers_config/default_controller: panda_simulator/p...
 * /controllers_config/position_controller: panda_simulator/p...
 * /controllers_config/torque_controller: panda_simulator/e...
 * /controllers_config/velocity_controller: panda_simulator/v...
 * /franka_gripper/robot_ip: sim
 * /gazebo/enable_ros_network: True
 * /gripper_config/default_grasp_epsilon/inner: 0.005
 * /gripper_config/default_grasp_epsilon/outer: 0.005
 * /gripper_config/default_speed: 0.1
 * /gripper_config/joint_names: ['panda_finger_jo...
 * /panda_simulator/effort_joint_gravity_controller/joints/panda_joint1_controller/joint: panda_joint1
 * /panda_simulator/effort_joint_gravity_controller/joints/panda_joint2_controller/joint: panda_joint2
 * /panda_simulator/effort_joint_gravity_controller/joints/panda_joint3_controller/joint: panda_joint3
 * /panda_simulator/effort_joint_gravity_controller/joints/panda_joint4_controller/joint: panda_joint4
 * /panda_simulator/effort_joint_gravity_controller/joints/panda_joint5_controller/joint: panda_joint5
 * /panda_simulator/effort_joint_gravity_controller/joints/panda_joint6_controller/joint: panda_joint6
 * /panda_simulator/effort_joint_gravity_controller/joints/panda_joint7_controller/joint: panda_joint7
 * /panda_simulator/effort_joint_gravity_controller/type: panda_sim_control...
 * /panda_simulator/effort_joint_torque_controller/joints/panda_joint1_controller/joint: panda_joint1
 * /panda_simulator/effort_joint_torque_controller/joints/panda_joint2_controller/joint: panda_joint2
 * /panda_simulator/effort_joint_torque_controller/joints/panda_joint3_controller/joint: panda_joint3
 * /panda_simulator/effort_joint_torque_controller/joints/panda_joint4_controller/joint: panda_joint4
 * /panda_simulator/effort_joint_torque_controller/joints/panda_joint5_controller/joint: panda_joint5
 * /panda_simulator/effort_joint_torque_controller/joints/panda_joint6_controller/joint: panda_joint6
 * /panda_simulator/effort_joint_torque_controller/joints/panda_joint7_controller/joint: panda_joint7
 * /panda_simulator/effort_joint_torque_controller/topic: /panda_simulator/...
 * /panda_simulator/effort_joint_torque_controller/type: panda_sim_control...
 * /panda_simulator/joint_state_controller/publish_rate: 100
 * /panda_simulator/joint_state_controller/type: joint_state_contr...
 * /panda_simulator/panda_gripper_controller/joints/panda_finger_joint1_controller/joint: panda_finger_joint1
 * /panda_simulator/panda_gripper_controller/joints/panda_finger_joint1_controller/pid/d: 0.5
 * /panda_simulator/panda_gripper_controller/joints/panda_finger_joint1_controller/pid/i: 50
 * /panda_simulator/panda_gripper_controller/joints/panda_finger_joint1_controller/pid/p: 5000
 * /panda_simulator/panda_gripper_controller/joints/panda_finger_joint2_controller/joint: panda_finger_joint2
 * /panda_simulator/panda_gripper_controller/joints/panda_finger_joint2_controller/pid/d: 0.5
 * /panda_simulator/panda_gripper_controller/joints/panda_finger_joint2_controller/pid/i: 50
 * /panda_simulator/panda_gripper_controller/joints/panda_finger_joint2_controller/pid/p: 5000
 * /panda_simulator/panda_gripper_controller/pub_topic: /franka_gripper/j...
 * /panda_simulator/panda_gripper_controller/type: panda_sim_control...
 * /panda_simulator/position_joint_position_controller/joints/panda_joint1_controller/joint: panda_joint1
 * /panda_simulator/position_joint_position_controller/joints/panda_joint1_controller/pid/d: 25.0
 * /panda_simulator/position_joint_position_controller/joints/panda_joint1_controller/pid/i: 10.0
 * /panda_simulator/position_joint_position_controller/joints/panda_joint1_controller/pid/p: 3000
 * /panda_simulator/position_joint_position_controller/joints/panda_joint2_controller/joint: panda_joint2
 * /panda_simulator/position_joint_position_controller/joints/panda_joint2_controller/pid/d: 30.0
 * /panda_simulator/position_joint_position_controller/joints/panda_joint2_controller/pid/i: 10.0
 * /panda_simulator/position_joint_position_controller/joints/panda_joint2_controller/pid/p: 3000
 * /panda_simulator/position_joint_position_controller/joints/panda_joint3_controller/joint: panda_joint3
 * /panda_simulator/position_joint_position_controller/joints/panda_joint3_controller/pid/d: 30.0
 * /panda_simulator/position_joint_position_controller/joints/panda_joint3_controller/pid/i: 10.0
[... 180 lines cut ...]
Started ['panda_simulator/panda_gripper_controller'] successfully
Started ['panda_simulator/position_joint_position_controller'] successfully
[ INFO] [1614020235.995781021, 0.579000000]: Initializing JointArrayController(type:/panda_simulator/effort_joint_torque_controller) with 7 joints.
[ INFO] [1614020235.995857193, 0.579000000]: Loading sub-controller 'panda_joint1_controller', Namespace: /panda_simulator/effort_joint_torque_controller/joints/panda_joint1_controller
[ INFO] [1614020236.000283847, 0.583000000]: Loading sub-controller 'panda_joint2_controller', Namespace: /panda_simulator/effort_joint_torque_controller/joints/panda_joint2_controller
[ INFO] [1614020236.002639880, 0.585000000]: Loading sub-controller 'panda_joint3_controller', Namespace: /panda_simulator/effort_joint_torque_controller/joints/panda_joint3_controller
[ INFO] [1614020236.005046177, 0.588000000]: Loading sub-controller 'panda_joint4_controller', Namespace: /panda_simulator/effort_joint_torque_controller/joints/panda_joint4_controller
[ INFO] [1614020236.006751958, 0.590000000]: Loading sub-controller 'panda_joint5_controller', Namespace: /panda_simulator/effort_joint_torque_controller/joints/panda_joint5_controller
[ INFO] [1614020236.008132262, 0.591000000]: Loading sub-controller 'panda_joint6_controller', Namespace: /panda_simulator/effort_joint_torque_controller/joints/panda_joint6_controller
[ INFO] [1614020236.009505021, 0.592000000]: Loading sub-controller 'panda_joint7_controller', Namespace: /panda_simulator/effort_joint_torque_controller/joints/panda_joint7_controller
Loaded 'panda_simulator/effort_joint_torque_controller'
Started ['panda_simulator/joint_state_controller'] successfully
[controller_spawner-7] process has finished cleanly
log file: /home/altair/.ros/log/c59f8780-753f-11eb-8ab4-3c6aa7b729ab/controller_spawner-7*.log
[controller_spawner_stopped-8] process has finished cleanly
log file: /home/altair/.ros/log/c59f8780-753f-11eb-8ab4-3c6aa7b729ab/controller_spawner_stopped-8*.log
[panda_gripper_controller_spawner_stopped-9] process has finished cleanly
log file: /home/altair/.ros/log/c59f8780-753f-11eb-8ab4-3c6aa7b729ab/panda_gripper_controller_spawner_stopped-9*.log
Segmentation fault (core dumped)
[ERROR] [1614020256.122124, 0.612000]: SetModelConfiguration service failed. Exiting.
[gazebo-2] process has died [pid 88725, exit code 139, cmd /opt/ros/melodic/lib/gazebo_ros/gzserver -e ode /home/altair/catkin_panda/src/panda_simulator/panda_gazebo/worlds/panda.world __name:=gazebo __log:=/home/altair/.ros/log/c59f8780-753f-11eb-8ab4-3c6aa7b729ab/gazebo-2.log].
log file: /home/altair/.ros/log/c59f8780-753f-11eb-8ab4-3c6aa7b729ab/gazebo-2*.log
[robot_description-4] process has died [pid 88735, exit code 1, cmd /opt/ros/melodic/lib/gazebo_ros/spawn_model -param robot_description -urdf -model panda -J panda::panda_joint1 0.000 -J panda::panda_joint2 -0.785 -J panda::panda_joint3 0.0 -J panda::panda_joint4 -2.356 -J panda::panda_joint5 0.0 -J panda::panda_joint6 1.57 -J panda::panda_joint7 0.785 __name:=robot_description __log:=/home/altair/.ros/log/c59f8780-753f-11eb-8ab4-3c6aa7b729ab/robot_description-4.log].
log file: /home/altair/.ros/log/c59f8780-753f-11eb-8ab4-3c6aa7b729ab/robot_description-4*.log
```

justagist, 2021-02-23:
Hi, I am not sure why but it looks like the gripper controller is not loading correctly. Did you change anything in the launch files or controller codes? Also, the moveit server should not be started till the simulation is fully loaded.

rouzinho, 2021-02-24:
Hi,

I did not modify anything, just followed the installation process. I actually do not have time to launch the moveit server since the segmentation fault happens before that.

justagist, 2021-02-24:
I am not sure what could be wrong as I can't replicate it. Can you try rebuilding the package in a clean workspace. If it still persists, please share the log file for the seg fault, and I will try to see what's going on.

rouzinho, 2021-03-04:
I did try multiple times with a clean install but always the same issue. I'm closing this issue because I found an alternate panda simulation which is working quite well for my needs. But anyway thanks a lot for the help !

closed 2021-03-04 by rouzinho