justagist/panda_simulator #28: Error: Moveit - Gazebo communication

StephanDieGeileSau, 2021-03-01:
When building the project like your tutorial, I get an error when executing the planned path in moveit.

`[ INFO] [1614583594.577708066, 101.569000000]: Execution request received
[ERROR] [1614583594.577888662, 101.569000000]: Unable to identify any set of controllers that can actuate the specified joints: [ panda_joint1 panda_joint2 panda_joint3 panda_joint4 panda_joint5 panda_joint6 panda_joint7 ]
[ERROR] [1614583594.577946947, 101.569000000]: Known controllers and their joints:`

Do I have to make changes in Gazebo first? or probably choose the right controller in the Moveit configs?

justagist, 2021-03-01:
You shouldn't have to make any changes in the config, and then demos should work out-of-the-box if you manage to build the package successfully.
Can you specify which commands you ran? Did you start the moveit server using `roslaunch panda_sim_moveit sim_move_group.launch` before running moveit demos? The error message seems to say the controllers are not being loaded correctly. Make sure all the runtime dependencies are met (`apt install ros-melodic-moveit ros-melodic-ros-control ros-melodic-joint-trajectory-controller`), and try running again.

StephanDieGeileSau, 2021-03-01:
I installed all dependencies specified in the dockerfile and also pip3 installed all requirements.txt libraries.

The launch order is following:
`roslaunch panda_gazebo panda_world.launch
roslaunch panda_sim_moveit sim_move_group.launch
roslaunch panda_simulator_examples demo_moveit.launch`

For the moveit server I get following Warning and Error:
`[ WARN] [1614599318.717904557, 375.179000000]: Waiting for position_joint_trajectory_controller/follow_joint_trajectory to come up
[ WARN] [1614599324.732847140, 381.179000000]: Waiting for position_joint_trajectory_controller/follow_joint_trajectory to come up
[ERROR] [1614599330.754851815, 387.179000000]: Action client not connected: position_joint_trajectory_controller/follow_joint_trajectory
[ WARN] [1614599335.853593910, 392.258000000]: Waiting for franka_gripper/gripper_action to come up
[ WARN] [1614599341.883159988, 398.258000000]: Waiting for franka_gripper/gripper_action to come up
[ERROR] [1614599347.913771236, 404.258000000]: Action client not connected: franka_gripper/gripper_action
`

When running the move_robot.py I get following Traceback:
`$ python move_robot.py 
Traceback (most recent call last):
  File "move_robot.py", line 4, in <module>
    from franka_core_msgs.msg import JointCommand, RobotState
ImportError: No module named franka_core_msgs.msg
`

justagist, 2021-03-01:
> I installed all dependencies specified in the dockerfile and also pip3 installed all requirements.txt libraries.

The package requires python 2, so `pip3 install` won't do it, you'll have to use the python 2 variant (`pip install` usually).

> For the moveit server I get following Warning and Error:

This seems to suggest that the simulator launch is not complete (the custom action servers for joint trajectory and gripper don't seem to have started). You should wait till the simulator is fully loaded before starting the moveit server. If the problem still persists, you might want to disable the automatic start of the custom action servers using the flag `use_custom_action_servers:=false` (`roslaunch panda_gazebo panda_world.launch use_custom_action_servers:=false`), and then start the action server nodes separately (see the nodes started by this flag in the launch file). Then start the moveit server.

> When running the move_robot.py I get following Traceback:
`$ python move_robot.py Traceback (most recent call last): File "move_robot.py", line 4, in <module> from franka_core_msgs.msg import JointCommand, RobotState ImportError: No module named franka_core_msgs.msg`

If you already have met the dependencies (including `franka_ros_interface` package), then it seems either the workspace is not built properly, or it is not sourced (`source <catkin_ws>/devel/setup.bash`).

StephanDieGeileSau, 2021-03-01:
We solved the issue by:

1. changing the url for the REST server API for Gazebo: [this issue](https://answers.gazebosim.org//question/22263/error-in-rest-request-for-accessing-apiignitionorg/)
2. reinstalling numpy etc. via: `python -m pip install numpy-quaternion`

Thank you very much!!!

justagist, 2021-03-01:
You hadn't mentioned the REST API error, or I could have pointed it out for you.
Anyway, glad to hear you got it working!

closed 2021-03-01 by justagist