justagist/panda_simulator #35: issue with the r.velocities()

xzhuzhu, 2021-04-13:
Thanks for your work, it helps me a lot. now I meet a problem, I changed the office code "cartesian_impedance_example_controller.cpp" to python but found the arm abnormal jitter 
in the gazebo,and then I print the r.velocities() and found no zero at start. Then I wrote a easy code      
   rospy.init_node("test_node")
    r = PandaArm()
    rate = rospy.Rate(400)    
    while not rospy.is_shutdown():
        print r.velocities()
but got the output
[ 2.39455761e-02  1.89977710e-03 -2.23809963e-02 -3.04818945e-02
  7.50835108e-02  3.56811108e-01  4.63242645e+00]
[ 2.39455761e-02  1.89977710e-03 -2.23809963e-02 -3.04818945e-02
  7.50835108e-02  3.56811108e-01  4.63242645e+00]
[ 2.39455761e-02  1.89977710e-03 -2.23809963e-02 -3.04818945e-02
  7.50835108e-02  3.56811108e-01  4.63242645e+00]
[ 0.06962769  0.00872245 -0.07593144 -0.03078833  0.05199004  0.36187647
  4.62442711]
[ 0.06962769  0.00872245 -0.07593144 -0.03078833  0.05199004  0.36187647
  4.62442711]
[ 0.06962769  0.00872245 -0.07593144 -0.03078833  0.05199004  0.36187647
  4.62442711]
[ 0.06962769  0.00872245 -0.07593144 -0.03078833  0.05199004  0.36187647
  4.62442711]
[ 0.06962769  0.00872245 -0.07593144 -0.03078833  0.05199004  0.36187647
  4.62442711]

How can I fix this?thank you~

justagist, 2021-04-13:
Hi,
There shouldn't be any jitter in gazebo (at least visually). Make sure your are using the latest version of [`franka_panda_description`](https://github.com/justagist/franka_panda_description) package. However, as you noticed, the velocities are not always zero, especially when the robot configuration is away from the neutral pose configuration.
I am aware of this issue, but unfortunately, I could not solve this problem yet. This has to do with the dynamics values of the robot description. Make sure your are using the latest version of `franka_panda_description` package, which has reasonably better dynamics parameters. This is related to the issue justagist/franka_panda_description#5 in that package. If you do manage to find better dynamics definitions, please do let me know. Any help is appreciated.

Also, the noisy values of velocities shouldn't matter too much when implementing a cartesian impedance controller, I think. You can try reducing the damping gains of the controller. The [`demo_task_space_control.launch`](https://github.com/justagist/panda_simulator/blob/melodic-devel/panda_simulator_examples/scripts/task_space_control_with_fri.py) example in this repository implements a simplified impedance controller (no null-space control, coriolis compensation etc. but these can be included easily) in python.

justagist, 2021-04-15:
> Hi,
> There shouldn't be any jitter in gazebo (at least visually). Make sure your are using the latest version of [`franka_panda_description`](https://github.com/justagist/franka_panda_description) [...]

This issue should now be (partially) fixed. Please use the latest version of the `franka_panda_description` package. The velocities are more realistic with the updated dynamics description.

closed 2021-04-26 by justagist

changanliushao, 2021-08-04:
I use the latest version of the franka_panda_description package, but the problem do not resolve
the output as below, 

[ 0.0209783  -0.01251312 -0.03375443 -0.02986879 -0.01649542  0.0286632
  0.00507145]
[ 0.0209783  -0.01251312 -0.03375443 -0.02986879 -0.01649542  0.0286632
  0.00507145]
[ 0.02101549 -0.01253853 -0.03370266 -0.02988156 -0.01646306  0.0286576
  0.00507724]
[ 0.02101549 -0.01253853 -0.03370266 -0.02988156 -0.01646306  0.0286576
  0.00507724]
[ 0.02101549 -0.01253853 -0.03370266 -0.02988156 -0.01646306  0.0286576
  0.00507724]
[ 0.02101549 -0.01253853 -0.03370266 -0.02988156 -0.01646306  0.0286576
  0.00507724]
[ 0.02101549 -0.01253853 -0.03370266 -0.02988156 -0.01646306  0.0286576
  0.00507724]

justagist, 2021-08-04:
Hi @changanliushao,
Unfortunately, that is currently the best model I have and it is not possible to get better dynamics measurements than what you showed with this model. 
If you are able to improve the model, it would be a great contribution.

changanliushao, 2021-08-05:
> Hi @changanliushao,
> Unfortunately, that is currently the best model I have and it is not possible to get better dynamics measurements than what you showed with this model.
> If you are able to improve the model, it would be a great contribution.

Thanks, I don't think I can improve this model. 
Maybe due to this problem, the trajectory is not satisfactory when I use API set_joint_velocities. but the data(joint_velocities) work well on real robot.

the code such as:
rospy.init_node("test_node")
r = PandaArm()
rate = rospy.Rate(1000)
while not rospy.is_shutdown():
    r.set_joint_velocities(joint_velocities)
    rate.sleep()