justagist/panda_simulator #36: const class KDL::Joint’ has no member named ‘getInertia

upuphero, 2021-04-14:
There is a problem when I catkin_make about this package 

`/home/ye/ws_move/src/panda_simulator/panda_gazebo/src/kdl_methods.cpp:` In member function ‘virtual int panda_gazebo::KDLMethods::RNECartToJnt(const KDL::JntArray&, const KDL::JntArray&, const KDL::JntArray&, const Wrenches&, KDL::JntArray&, KDL::Twist)’:
/home/ye/finalyearproject/src/panda_simulator/panda_gazebo/src/kdl_methods.cpp:119:61: error: ‘const class KDL::Joint’ has no member named ‘getInertia’; did you mean ‘inertia’?
                 torques(j)+=chain_.getSegment(i).getJoint().getInertia()*q_dotdot(j);  // add torque from joint inertia
                                                                                      ^~~~~~~~~~
                                                                                      inertia
/home/ye/finalyearproject/src/panda_simulator/panda_gazebo/src/kdl_methods.cpp: In member function ‘virtual int panda_gazebo::KDLMethods::JntToMass(const KDL::JntArray&, KDL::JntSpaceInertiaMatrix&)’:
/home/ye/finalyearproject/src/panda_simulator/panda_gazebo/src/kdl_methods.cpp:227:51: error: ‘const class KDL::Joint’ has no member named ‘getInertia’; did you mean ‘inertia’?
           H(k,k)+=chain_.getSegment(i).getJoint().getInertia();  // add joint inertia
                                                                           ^~~~~~~~~~
                                                                            inertia
panda_simulator/panda_gazebo/CMakeFiles/panda_gazebo_ros_control.dir/build.make:134: recipe for target 'panda_simulator/panda_gazebo/CMakeFiles/panda_gazebo_ros_control.dir/src/kdl_methods.cpp.o' failed`

I am trying to figure out what is wrong with this inertia? And if I replace the  getInertia with inertia  but it shows that 
`opt/ros/melodic/include/kdl/joint.hpp:202:16: note: declared private here`

justagist, 2021-04-14:
You have to build a specific version of `orocos_kinematics_dynamics` package as mentioned in the readme.
```bash
cd orocos_kinematics_dynamics && git checkout b35c424e77ebc5b7e6f1c5e5c34f8a4666fbf5bc
```

upuphero, 2021-04-14:
Thanks a lot!!! I will try when the power is back

closed 2021-06-22 by justagist