justagist/panda_simulator #37: ImportError: No module named quaternion [joint_trajectory_server_emulator-11] process has died

upuphero, 2021-04-15:
I am facing another problem that when I run the command `roslaunch panda_gazebo panda_world.launch`
, the terminal will report `ImportError: No module named quaternion
[joint_trajectory_server_emulator-11] process has died [pid 15370, exit code 1, cmd /home/ye/finalyearproject/src/panda_simulator/panda_sim_custom_action_server/scripts/start_joint_trajectory_server.py __name:=joint_trajectory_server_emulator __log:=/home/ye/.ros/log/ec2ff2f2-9d96-11eb-9889-000c294926c2/joint_trajectory_server_emulator-11.log].
log file: /home/ye/.ros/log/ec2ff2f2-9d96-11eb-9889-000c294926c2/joint_trajectory_server_emulator-11*.log`

When I run the command `roslaunch panda_simulator_examples demo_moveit.launch`, it will report
`[rviz_ubuntu_14368_8967066934926071941-15] process has died [pid 14434, exit code 127, cmd /opt/ros/melodic/lib/rviz/rviz -d /home/ye/finalyearproject/src/franka_ros_interface/franka_moveit/config/moveit_demo.rviz __name:=rviz_ubuntu_14368_8967066934926071941 __log:=/home/ye/.ros/log/6bd8cce6-9d96-11eb-9889-000c294926c2/rviz_ubuntu_14368_8967066934926071941-15.log].
log file: /home/ye/.ros/log/6bd8cce6-9d96-11eb-9889-000c294926c2/rviz_ubuntu_14368_8967066934926071941-15*.log` and the window of rviz disappeared as soon as it appears. Now I could see the robot model in gazebo but I could not do anything

I have catkin build all the workspace and it succeeded. But there is three ignored packages. Maybe the building of  franka_ros_interface is wrong but it doesn't tell anything from catkin build

`ye@ubuntu:~/finalyearproject$ catkin build
--------------------------------------------------------------
Profile:                     default
Extending:          [cached] /opt/ros/melodic
Workspace:                   /home/ye/finalyearproject
--------------------------------------------------------------
Build Space:        [exists] /home/ye/finalyearproject/build
Devel Space:        [exists] /home/ye/finalyearproject/devel
Install Space:      [unused] /home/ye/finalyearproject/install
Log Space:          [exists] /home/ye/finalyearproject/logs
Source Space:       [exists] /home/ye/finalyearproject/src
DESTDIR:            [unused] None
--------------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
--------------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
--------------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
--------------------------------------------------------------
Workspace configuration appears valid.
--------------------------------------------------------------
[build] Found '22' packages in 0.0 seconds.                                                                            
[build] Updating package table.                                                                                        
Starting  >>> communication                                                                                            
Starting  >>> franka_core_msgs                                                                                         
Starting  >>> franka_moveit                                                                                            
Starting  >>> franka_panda_description                                                                                 
Starting  >>> fyp                                                                                                      
Starting  >>> orocos_kdl                                                                                               
Finished  <<< fyp                                           [ 0.1 seconds ]                                            
Starting  >>> panda_hardware_interface                                                                                 
Finished  <<< franka_panda_description                      [ 0.1 seconds ]                                            
Starting  >>> panda_sim_moveit                                                                                         
Finished  <<< panda_hardware_interface                      [ 0.1 seconds ]                                            
Starting  >>> push                                                                                                     
Finished  <<< orocos_kdl                                    [ 0.2 seconds ]                                            
Finished  <<< panda_sim_moveit                              [ 0.1 seconds ]                                            
Finished  <<< communication                                 [ 0.5 seconds ]                                            
Starting  >>> turtlebot3_fake                                                                                          
Starting  >>> turtlebot3_gazebo                                                                                        
Starting  >>> python_orocos_kdl                                                                                        
Finished  <<< push                                          [ 0.1 seconds ]                                            
Finished  <<< python_orocos_kdl                             [ 0.1 seconds ]                                            
Finished  <<< franka_moveit                                 [ 0.9 seconds ]                                            
Finished  <<< turtlebot3_fake                               [ 0.4 seconds ]                                            
Finished  <<< turtlebot3_gazebo                             [ 0.5 seconds ]                                            
Finished  <<< franka_core_msgs                              [ 1.6 seconds ]                                            
Starting  >>> franka_interface                                                                                         
Starting  >>> franka_ros_controllers                                                                                   
Starting  >>> panda_sim_controllers                                                                                    
Finished  <<< panda_sim_controllers                         [ 7.2 seconds ]                                            
Finished  <<< franka_ros_controllers                        [ 15.4 seconds ]                                           
Finished  <<< franka_interface                              [ 20.1 seconds ]                                           
Starting  >>> franka_tools                                                                                             
Starting  >>> panda_sim_custom_action_server                                                                           
Finished  <<< panda_sim_custom_action_server                [ 0.1 seconds ]                                            
Finished  <<< franka_tools                                  [ 1.1 seconds ]                                            
Starting  >>> panda_gazebo                                                                                             
Finished  <<< panda_gazebo                                  [ 8.5 seconds ]                                            
Starting  >>> panda_simulator_examples                                                                                 
Finished  <<< panda_simulator_examples                      [ 0.1 seconds ]                                            
[build] Summary: All 19 packages succeeded!                                                                            
[build]   Ignored:   3 packages were skipped or are blacklisted.                                                       
[build]   Warnings:  None.                                                                                             
[build]   Abandoned: None.                                                                                             
[build]   Failed:    None.                                                                                             
[build] Runtime: 31.7 seconds total.`

justagist, 2021-04-15:
> the terminal will report ImportError: No module named quaternion [joint_trajectory_server_emulator-11] process has died [pid 15370, exit code 1

This looks like you haven't installed the quaternion package. Please follow the installation instructions carefully: `pip install numpy numpy-quaternion` (you may need `sudo`).
Also, make sure you start the moveit server before running the moveit demo (follow instructions in the README for running the demo).

CodeFinder2, 2021-05-20:
> [...]
> This looks like you haven't installed the quaternion package. Please follow the installation instructions carefully: `pip install numpy numpy-quaternion` (you may need `sudo`).
> Also, make sure [...]

The problem with this is: `numpy` also gets installed as a system package along with various ROS (system) packages. Installing it with pip(3) **and**  using `sudo` is [**definitely not** a good idea](https://stackoverflow.com/a/22517157).

Only installing `numpy-quaternion` via pip (`pip3 install --user numpy-quaternion` because there is no system package available for Noetic / Ubuntu 20.04) causes this error:
```shell
RuntimeError: module compiled against API version 0xe but this version of numpy is 0xd
Traceback (most recent call last):
  File "/hidden/catkin_ws/devel/lib/panda_sim_custom_action_server/start_gripper_action_server.py", line 15, in <module>
    exec(compile(fh.read(), python_script, 'exec'), context)
  File "/hidden/catkin_ws/src/panda_simulator/panda_sim_custom_action_server/scripts/start_gripper_action_server.py", line 31, in <module>
    from panda_sim_custom_action_server import GripperActionServer
  File "/hidden/catkin_ws/devel/lib/python3/dist-packages/panda_sim_custom_action_server/__init__.py", line 34, in <module>
    exec(__fh.read())
  File "<string>", line 18, in <module>
  File "/hidden/catkin_ws/src/panda_simulator/panda_sim_custom_action_server/src/panda_sim_custom_action_server/joint_trajectory_action_server.py", line 56, in <module>
    import franka_interface
  File "/hidden/catkin_ws/src/franka_ros_interface/franka_interface/src/franka_interface/__init__.py", line 18, in <module>
    from .arm import ArmInterface
  File "/hidden/catkin_ws/src/franka_ros_interface/franka_interface/src/franka_interface/arm.py", line 39, in <module>
    import quaternion
  File "/home/abo/.local/lib/python3.8/site-packages/quaternion/__init__.py", line 25, in <module>
    from .numpy_quaternion import (
ImportError: numpy.core.multiarray failed to import
```
I think this is caused by the fact that the system comes with a `numpy` version that is different from the one that is being used by `numpy-quaternion` (but not 100% sure).

Any better ideas? (I am having the same / a very similiar problem...)

justagist, 2021-05-20:
Hi @CodeFinder2, thanks for this!
 
I agree that using `sudo` is probably not safe for `pip` commands, but `quaternion` and `numpy` are pretty standard packages and I would assume they would be safe to install using `sudo`. But you'd be right if you say this is not a safe assumption.

> I think this is caused by the fact that the system comes with a numpy version that is different from the one that is being used by numpy-quaternion (but not 100% sure).

Yes, I would have came to the same conclusion, and would have suggested upgrading the installed `numpy` package and trying to install `quaternion` again, as you have done. 

Thanks anyway for your detailed solution.
Did you have to install `python-is-python3` package to solve this issue, or was this only to be safe?

CodeFinder2, 2021-05-20:
`python-is-python3' is not necessary. Calling `python3` explicitly would work as well. 🙂

closed 2021-06-22 by justagist