justagist/panda_simulator #40: Moveit Demo

upuphero, 2021-04-23:
When running the Moveit Demo, rviz came out but disappeared quickly. I have run the Task-space control using Franka ROS Interface successfully.
I am sure I follow the Usage and I type these three commands in order in terminals but still find the bugs
`roslaunch panda_gazebo panda_world.launch`
`roslaunch panda_sim_moveit sim_move_group.launch`
`roslaunch panda_simulator_examples demo_moveit.launch`

The bug shows below:
`[rviz_ubuntu_4376_5041817988713892492-1] process has died [pid 4525, exit code 127, cmd /opt/ros/melodic/lib/rviz/rviz -d /home/ye/ws_moveit/src/franka_ros_interface/franka_moveit/config/moveit_demo.rviz __name:=rviz_ubuntu_4376_5041817988713892492 __log:=/home/ye/.ros/log/142905d8-a403-11eb-aeb0-000c294926c2/rviz_ubuntu_4376_5041817988713892492-1.log].
log file: /home/ye/.ros/log/142905d8-a403-11eb-aeb0-000c294926c2/rviz_ubuntu_4376_5041817988713892492-1*.log`

justagist, 2021-04-23:
Hi, these messages don't really help. There must be some other warning or error messages before; if not, please look into the log files and let me know what you find.

upuphero, 2021-04-23:
> Hi, these messages don't really help. There must be some other warning or error messages before; if not, please look into the log files and let me know what you find.

I found the log folder dfba3fcc-a409-11eb-aeb0-000c294926c2， but I can't find the  rviz_ubuntu_121048_1056280393894983898-1*.log

Here is the bug
`[rviz_ubuntu_121048_1056280393894983898-1] process has died [pid 121190, exit code 127, cmd /opt/ros/melodic/lib/rviz/rviz -d /home/ye/ws_moveit/src/franka_ros_interface/franka_moveit/config/moveit_demo.rviz __name:=rviz_ubuntu_121048_1056280393894983898 __log:=/home/ye/.ros/log/dfba3fcc-a409-11eb-aeb0-000c294926c2/rviz_ubuntu_121048_1056280393894983898-1.log].
log file: /home/ye/.ros/log/dfba3fcc-a409-11eb-aeb0-000c294926c2/rviz_ubuntu_121048_1056280393894983898-1*.log
`

I copied all the sentence in the terminal

roslaunch panda_simulator_examples demo_moveit.launch 
... logging to /home/ye/.ros/log/dfba3fcc-a409-11eb-aeb0-000c294926c2/roslaunch-ubuntu-121048.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ubuntu:37411/

PARAMETERS
 * /panda_arm/kinematics_solver: kdl_kinematics_pl...
 * /panda_arm/kinematics_solver_search_resolution: 0.005
 * /panda_arm/kinematics_solver_timeout: 0.05
 * /robot_description_kinematics/panda_arm/kinematics_solver: kdl_kinematics_pl...
 * /robot_description_kinematics/panda_arm/kinematics_solver_search_resolution: 0.005
 * /robot_description_kinematics/panda_arm/kinematics_solver_timeout: 0.05
 * /robot_description_planning/joint_limits/panda_finger_joint1/has_acceleration_limits: False
 * /robot_description_planning/joint_limits/panda_finger_joint1/has_velocity_limits: True
 * /robot_description_planning/joint_limits/panda_finger_joint1/max_acceleration: 0
 * /robot_description_planning/joint_limits/panda_finger_joint1/max_velocity: 0.1
 * /robot_description_planning/joint_limits/panda_finger_joint2/has_acceleration_limits: False
 * /robot_description_planning/joint_limits/panda_finger_joint2/has_velocity_limits: True
 * /robot_description_planning/joint_limits/panda_finger_joint2/max_acceleration: 0
 * /robot_description_planning/joint_limits/panda_finger_joint2/max_velocity: 0.1
 * /robot_description_planning/joint_limits/panda_joint1/has_acceleration_limits: True
 * /robot_description_planning/joint_limits/panda_joint1/has_velocity_limits: True
 * /robot_description_planning/joint_limits/panda_joint1/max_acceleration: 3.75
 * /robot_description_planning/joint_limits/panda_joint1/max_velocity: 2.175
 * /robot_description_planning/joint_limits/panda_joint2/has_acceleration_limits: True
 * /robot_description_planning/joint_limits/panda_joint2/has_velocity_limits: True
 * /robot_description_planning/joint_limits/panda_joint2/max_acceleration: 1.875
 * /robot_description_planning/joint_limits/panda_joint2/max_velocity: 2.175
 * /robot_description_planning/joint_limits/panda_joint3/has_acceleration_limits: True
 * /robot_description_planning/joint_limits/panda_joint3/has_velocity_limits: True
 * /robot_description_planning/joint_limits/panda_joint3/max_acceleration: 2.5
 * /robot_description_planning/joint_limits/panda_joint3/max_velocity: 2.175
 * /robot_description_planning/joint_limits/panda_joint4/has_acceleration_limits: True
 * /robot_description_planning/joint_limits/panda_joint4/has_velocity_limits: True
 * /robot_description_planning/joint_limits/panda_joint4/max_acceleration: 3.125
 * /robot_description_planning/joint_limits/panda_joint4/max_velocity: 2.175
 * /robot_description_planning/joint_limits/panda_joint5/has_acceleration_limits: True
 * /robot_description_planning/joint_limits/panda_joint5/has_velocity_limits: True
 * /robot_description_planning/joint_limits/panda_joint5/max_acceleration: 3.75
 * /robot_description_planning/joint_limits/panda_joint5/max_velocity: 2.61
 * /robot_description_planning/joint_limits/panda_joint6/has_acceleration_limits: True
 * /robot_description_planning/joint_limits/panda_joint6/has_velocity_limits: True
 * /robot_description_planning/joint_limits/panda_joint6/max_acceleration: 5
 * /robot_description_planning/joint_limits/panda_joint6/max_velocity: 2.61
 * /robot_description_planning/joint_limits/panda_joint7/has_acceleration_limits: True
 * /robot_description_planning/joint_limits/panda_joint7/has_velocity_limits: True
 * /robot_description_planning/joint_limits/panda_joint7/max_acceleration: 5
 * /robot_description_planning/joint_limits/panda_joint7/max_velocity: 2.61
 * /robot_description_semantic: <?xml version="1....
 * /rosdistro: melodic
 * /rosversion: 1.14.10

NODES
  /
    demo_boxes_scene_loader (franka_moveit/test_scene_interface.py)
    rviz_ubuntu_121048_1056280393894983898 (rviz/rviz)
[... 23 lines cut ...]
may wish to install numba.
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!

  warnings.warn(warning_text)
============ Starting test ...
[ INFO] [1619165776.346635124, 943.286000000]: Loading robot model 'panda'...
[ WARN] [1619165776.346797683, 943.286000000]: Skipping virtual joint 'virtual_joint' because its child frame 'panda_link0' does not match the URDF frame 'world'
[ INFO] [1619165776.347384292, 943.287000000]: No root/virtual joint specified in SRDF. Assuming fixed joint
[ INFO] [1619165776.356541166, 943.295000000]: Loading robot model 'panda'...
[... cut ...]
[rviz_ubuntu_121048_1056280393894983898-1] process has died [pid 121190, exit code 127, cmd /opt/ros/melodic/lib/rviz/rviz -d /home/ye/ws_moveit/src/franka_ros_interface/franka_moveit/config/moveit_demo.rviz __name:=rviz_ubuntu_121048_1056280393894983898 __log:=/home/ye/.ros/log/dfba3fcc-a409-11eb-aeb0-000c294926c2/rviz_ubuntu_121048_1056280393894983898-1.log].
log file: /home/ye/.ros/log/dfba3fcc-a409-11eb-aeb0-000c294926c2/rviz_ubuntu_121048_1056280393894983898-1*.log
============ Press `Enter` to begin the tutorial by setting up the scene_interface (press ctrl-d to exit) ...

If there is anything else helpful, I am glad to supply. Thanks a lot!

justagist, 2021-04-23:
> /opt/ros/melodic/lib/rviz/rviz: symbol lookup error: /opt/ros/melodic/lib/libmoveit_dynamics_solver.so.1.0.7: undefined symbol: _ZNK3KDL6Vector4NormEv

This suggests that you are not using the specified version of `orocos_kinematics_dynamics` package. Please checkout to the right commit version of the package after cloning it into your workspace as I told you [here](https://github.com/justagist/panda_simulator/issues/36#issuecomment-819619587). Then make sure you do a clean build of your workspace (delete the `build` and `devel` directories in your workspace, and run `catkin build` again).

closed 2021-06-22 by justagist