justagist/panda_simulator #42: I use cpp to imitate python script(move_robot.py). message md5sum is mismatch.

TracyLestrade, 2021-04-26:
I copied some .msg files from `franka_ros_interface\franka_common\franka_core_msgs\msg`  to my package(mypack) `mypack\msg`. 
and I change package.xml, CMakeList.txt to configure it.
it occured error. because it missed some dependencies. 
so, I copied `franka_ros\franka_msgs` this package to my catkin workspace.
it.  after that, I added package dependency.
`  
<build_depend>franka_msgs</build_depend>
  <exec_depend>franka_msgs</exec_depend>
`
now, compile successed. but I feel like some generated header is odd.
beside some message header. there are some `setcartesianimpedance.h seteeframe.h setkframe.h` etc. 
i quiet sure about my cpp code correct. and i ran my node get errors. goes with md5sum is mismatch.
md5:
JointCommand:766f72e801d6180a86db23dbf2c2a56f
RobotState:5b1c24f59317c607d71ccadfc0800957
how can i generate right header.  could u give me some advice? I also want to know the right message md5.

justagist, 2021-04-26:
You shouldn't have to copy the `msg` directories to your package. You should be able to use them directly in your package just by adding them as dependencies in your build configuration files. I believe the mismatch in the signatures could be because there are multiple instances of the same message files in your workspace.
Also, I will not be able to help you with building a different ROS package, sorry; please use other sources for figuring out these issues.
Closing this issue as it is not related to running or using this package. Please feel free to re-open if you think this wrong.

closed 2021-04-26 by justagist

TracyLestrade, 2021-04-27:
I checked.  the original md5 of _RobotState.py(franka_ros_interface) is same with mine. which it means the msg generation is fine. 
not multiple instances issue.
I guess there are some issue on my code. topic string I guess.