justagist/panda_simulator #48: Action client not connected. No controllers.

dennisushi, 2021-06-09:
Hi, i get the following when running the first demo. Gazebo starts but moveit plan and execute fail. Space control demo starts but I am not sure what I am supposed to see, it's just a robot spawning. 

```bash
[ WARN] [1623246837.082888209, 265.561000000]: Waiting for panda_arm_controller/follow_joint_trajectory to come up
[ WARN] [1623246843.260763153, 271.563000000]: Waiting for panda_arm_controller/follow_joint_trajectory to come up
[ERROR] [1623246849.457987399, 277.566000000]: Action client not connected: panda_arm_controller/follow_joint_trajectory
[ WARN] [1623246854.775775550, 282.669000000]: Waiting for panda_hand_controller/follow_joint_trajectory to come up
[ WARN] [1623246860.903231926, 288.669000000]: Waiting for panda_hand_controller/follow_joint_trajectory to come up
[ERROR] [1623246867.083942452, 294.673000000]: Action client not connected: panda_hand_controller/follow_joint_trajectory
```

After moveit plan or plan and execute:

```bash
[ERROR] [1623246874.713589285, 302.012000000]: Unable to identify any set of controllers that can actuate the specified joints: [ panda_joint1 panda_joint2 panda_joint3 panda_joint4 panda_joint5 panda_joint6 panda_joint7 ]
[ERROR] [1623246874.713613460, 302.012000000]: Known controllers and their joints:

[ERROR] [1623246874.713647697, 302.012000000]: Apparently trajectory initialization failed
```

dennisushi, 2021-06-09:
Not sure if related or intended, but in the gazebo world terminal one of the last lines it : 

```
[ INFO] [1623247242.074680821, 1.079000000]: ArmControllerInterface: Controller panda_simulator/position_joint_position_controller started; Controllers panda_simulator/effort_joint_torque_controller, panda_simulator/velocity_joint_velocity_controller stopped
```

justagist, 2021-06-09:
Hi,

> [ WARN] [1623246837.082888209, 265.561000000]: Waiting for panda_arm_controller/follow_joint_trajectory to come up

It looks like the wrong controller is being loaded. There are no controllers defined in this package with the namespace 'panda_arm_controller' or 'panda_hand_controller'. Have you modified the launch files or config files?

justagist, 2021-08-19:
Closing due to inactivity.

closed 2021-08-19 by justagist