justagist/panda_simulator #53: Process finished on startup and demo fail

rouzinho, 2021-08-24:
Hi,

I'm coming back with a fresh install (Pop Os - ubuntu 20-04) with ros noetic. The build did go well but I'm unable to launch the demo.
In the first terminal (roslaunch panda_gazebo panda_world.launch) :

```
roslaunch panda_gazebo panda_world.launch            
... logging to /home/altair/.ros/log/0bbe3e08-0516-11ec-b610-fb6eaaf7b86e/roslaunch-pop-os-97470.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: in-order processing became default in ROS Melodic. You can drop the option.
started roslaunch server http://pop-os:42933/

PARAMETERS
 * /SIMULATOR_: True
 * /arm/gravity_tip_name: panda_hand
 * /arm/root_name: panda_link0
 * /arm/tip_name: panda_hand
 * /controllers_config/default_controller: panda_simulator/p...
 * /controllers_config/position_controller: panda_simulator/p...
 * /controllers_config/torque_controller: panda_simulator/e...
 * /controllers_config/velocity_controller: panda_simulator/v...
 * /franka_gripper/robot_ip: sim
 * /gazebo/enable_ros_network: True
 * /gripper_config/default_grasp_epsilon/inner: 0.005
 * /gripper_config/default_grasp_epsilon/outer: 0.005
 * /gripper_config/default_speed: 0.1
 * /gripper_config/joint_names: ['panda_finger_jo...
 * /panda_simulator/effort_joint_gravity_controller/joints/panda_joint1_controller/joint: panda_joint1
 * /panda_simulator/effort_joint_gravity_controller/joints/panda_joint2_controller/joint: panda_joint2
 * /panda_simulator/effort_joint_gravity_controller/joints/panda_joint3_controller/joint: panda_joint3
 * /panda_simulator/effort_joint_gravity_controller/joints/panda_joint4_controller/joint: panda_joint4
 * /panda_simulator/effort_joint_gravity_controller/joints/panda_joint5_controller/joint: panda_joint5
 * /panda_simulator/effort_joint_gravity_controller/joints/panda_joint6_controller/joint: panda_joint6
 * /panda_simulator/effort_joint_gravity_controller/joints/panda_joint7_controller/joint: panda_joint7
 * /panda_simulator/effort_joint_gravity_controller/type: panda_sim_control...
 * /panda_simulator/effort_joint_torque_controller/joints/panda_joint1_controller/joint: panda_joint1
 * /panda_simulator/effort_joint_torque_controller/joints/panda_joint2_controller/joint: panda_joint2
 * /panda_simulator/effort_joint_torque_controller/joints/panda_joint3_controller/joint: panda_joint3
 * /panda_simulator/effort_joint_torque_controller/joints/panda_joint4_controller/joint: panda_joint4
 * /panda_simulator/effort_joint_torque_controller/joints/panda_joint5_controller/joint: panda_joint5
 * /panda_simulator/effort_joint_torque_controller/joints/panda_joint6_controller/joint: panda_joint6
 * /panda_simulator/effort_joint_torque_controller/joints/panda_joint7_controller/joint: panda_joint7
 * /panda_simulator/effort_joint_torque_controller/topic: /panda_simulator/...
 * /panda_simulator/effort_joint_torque_controller/type: panda_sim_control...
 * /panda_simulator/joint_state_controller/publish_rate: 100
 * /panda_simulator/joint_state_controller/type: joint_state_contr...
 * /panda_simulator/panda_gripper_controller/joints/panda_finger_joint1_controller/joint: panda_finger_joint1
 * /panda_simulator/panda_gripper_controller/joints/panda_finger_joint1_controller/pid/d: 0.5
 * /panda_simulator/panda_gripper_controller/joints/panda_finger_joint1_controller/pid/i: 50
 * /panda_simulator/panda_gripper_controller/joints/panda_finger_joint1_controller/pid/p: 5000
 * /panda_simulator/panda_gripper_controller/joints/panda_finger_joint2_controller/joint: panda_finger_joint2
 * /panda_simulator/panda_gripper_controller/joints/panda_finger_joint2_controller/pid/d: 0.5
 * /panda_simulator/panda_gripper_controller/joints/panda_finger_joint2_controller/pid/i: 50
 * /panda_simulator/panda_gripper_controller/joints/panda_finger_joint2_controller/pid/p: 5000
 * /panda_simulator/panda_gripper_controller/pub_topic: /franka_gripper/j...
 * /panda_simulator/panda_gripper_controller/type: panda_sim_control...
 * /panda_simulator/position_joint_position_controller/joints/panda_joint1_controller/joint: panda_joint1
 * /panda_simulator/position_joint_position_controller/joints/panda_joint1_controller/pid/d: 25.0
 * /panda_simulator/position_joint_position_controller/joints/panda_joint1_controller/pid/i: 10.0
 * /panda_simulator/position_joint_position_controller/joints/panda_joint1_controller/pid/p: 3000
 * /panda_simulator/position_joint_position_controller/joints/panda_joint2_controller/joint: panda_joint2
 * /panda_simulator/position_joint_position_controller/joints/panda_joint2_controller/pid/d: 30.0
 * /panda_simulator/position_joint_position_controller/joints/panda_joint2_controller/pid/i: 10.0
 * /panda_simulator/position_joint_position_controller/joints/panda_joint2_controller/pid/p: 3000
 * /panda_simulator/position_joint_position_controller/joints/panda_joint3_controller/joint: panda_joint3
 * /panda_simulator/position_joint_position_controller/joints/panda_joint3_controller/pid/d: 30.0
 * /panda_simulator/position_joint_position_controller/joints/panda_joint3_controller/pid/i: 10.0
[... 280 lines cut ...]
[ INFO] [1629835322.695440728, 40.083000000]: OpenGL device: NVIDIA GeForce RTX 3060 Laptop GPU/PCIe/SSE2
[ INFO] [1629835322.695492104, 40.083000000]: OpenGl version: 4.6 (GLSL 4.6).
============ Starting test ...
[ INFO] [1629835326.043801971, 43.381000000]: Loading robot model 'panda'...
[ INFO] [1629835326.043862784, 43.381000000]: Loading robot model 'panda'...
[ WARN] [1629835326.043889975, 43.381000000]: Skipping virtual joint 'virtual_joint' because its child frame 'panda_link0' does not match the URDF frame 'world'
[ INFO] [1629835326.043915623, 43.381000000]: No root/virtual joint specified in SRDF. Assuming fixed joint
[ WARN] [1629835326.043940109, 43.381000000]: Skipping virtual joint 'virtual_joint' because its child frame 'panda_link0' does not match the URDF frame 'world'
[ INFO] [1629835326.043971437, 43.381000000]: No root/virtual joint specified in SRDF. Assuming fixed joint
[ INFO] [1629835326.163202442, 43.498000000]: Starting planning scene monitor
[ INFO] [1629835326.168657447, 43.504000000]: Listening to '/planning_scene'
[ INFO] [1629835326.531438524, 43.865000000]: Starting planning scene monitor
[ INFO] [1629835326.531972379, 43.866000000]: waitForService: Service [/get_planning_scene] has not been advertised, waiting...
[ INFO] [1629835326.533571499, 43.868000000]: Listening to '/move_group/monitored_planning_scene'
[ INFO] [1629835326.535292546, 43.869000000]: waitForService: Service [/get_planning_scene] has not been advertised, waiting...
============ Press `Enter` to begin the tutorial by setting up the scene_interface (press ctrl-d to exit) ...[ INFO] [1629835331.539019239, 48.810000000]: Failed to call service get_planning_scene, have you launched move_group or called psm.providePlanningSceneService()?
[ INFO] [1629835331.542013788, 48.813000000]: Failed to call service get_planning_scene, have you launched move_group or called psm.providePlanningSceneService()?
[ INFO] [1629835331.542183665, 48.813000000]: Constructing new MoveGroup connection for group 'panda_arm' in namespace ''
[ERROR] [1629835361.552563965, 78.290000000]: Unable to connect to move_group action server 'move_group' within allotted time (30s)
[ERROR] [1629835361.557176535, 78.295000000]: Could not find parameter for database plugin name
```

I think the controllers are not properly starting but not sure at all...

Anway, thanks a lot for your work. That would save the end of my PhD if it works !

justagist, 2021-08-25:
This is probably because the moveit server is not running:
As mentioned in the readme, moveit server should be started manually before you can use any moveit functionalities. The error message also says that the move_group is not loaded. So the order of commands should be:
```bash
$ roslaunch panda_gazebo panda_world.launch
$ roslaunch panda_sim_moveit sim_move_group.launch # start moveit server
$ roslaunch panda_simulator_examples demo_moveit.launch # now you should be able to use moveit
```

rouzinho, 2021-08-25:
Thanks a lot. Somehow I was still reading the melodic readme...

closed 2021-08-25 by rouzinho

mitsui29, 2024-02-22:
Hey, i wanna ask, did u solve this issue? I'm working on this simulator. But i met this kind of questions too. (20.04. Noetic). Thanks in advance

rouzinho, 2024-02-22:
Yes, it's been a while I didn't use it but everything went well. I think I did as advised.

mitsui29, 2024-02-23:
> Yes, it's been a while I didn't use it but everything went well. I think I did as advised.

I already solved them. Just installed the right version of panda_moveit_config from another repo. Thanks for your time! Btw, do u know, where can i install the panda_robot pkg? just also under the same path with panda_simulator(/franka_ws/src)? I took a quick look at the code for panda_robot,  could not find more things apart from the intuitive interface classes and some low level control. I just wanna write some controllers(Cpp) that i derived from some papers. I think It is not necessary to download this package. It's enough to use panda_simulator and gazebo to finish my ideas. Maybe,  do you have some suggestions for me? Thanks a lot! :)