justagist/panda_simulator #54: Trajectory replacing support

petrvancjr, 2021-08-30:
Hi,

regarding your Trajectory Action controller. I wanted to ask if it supports the replace of the trajectory on go without the need of stopping the execution. Replace the old trajectory by aligned new one.

If not, what it possibly needed to be done to make it possible.

It would help us a lot. Thank you for your time.

justagist, 2021-08-30:
Hi,
This is not possible with the current formulation of the trajectory controller in this package. The controller provided is simply emulating the behaviour of a simplified [ROS Joint Trajectory Controller](http://wiki.ros.org/joint_trajectory_controller) using the `franka_ros_interface` developed for controlling the panda robot.
It looks like in the newer versions of ROS Joint Trajectory Controller, there is an option for [trajectory replacement](http://wiki.ros.org/joint_trajectory_controller/UnderstandingTrajectoryReplacement); I am not sure this does what you want. I am also not sure this will work with the custom action server defined in this package (I have not tested) because it was not originally designed to do this. 
Also, note that you won't be able to load ROS's joint trajectory controller directly in this simulator because it conflicts with the custom hardware interface defined for this project. So to support ROS joint trajectory action instead of the custom joint trajectory, you will either have to load default hardware instead of the custom hardware in the [robot description file](https://github.com/justagist/franka_panda_description/blob/62335618e7f141755524836c738efe9c01396440/robots/panda.transmission.xacro#L6-L26) and test it, or use a simpler simulator that uses default hardware interfaces and controllers such as [panda_simulation](https://github.com/erdalpekel/panda_simulation) and test if the 'trajectory replacement' feature is what you want. Alternatively, you can directly test in on the real robot as it loads ROS joint trajectory action by default if you are using `franka_ros` or `franka_ros_interface`.

petrvancjr, 2021-08-31:
When I tried to replace the trajectory, I observed an outage in trajectory tracing (row 3, row 4).
The attached plot:
- Series: **red - sent trajectory** to the controller, **blue - real trajectory**
- Figures: **rows** (1,2,3,4) - trajectory replacement iteration; **columns** (1,2) - **positions**, **velocities**

[image]

Even when I tried to remove the execution stop, I didn't observe any change. This is the reason that the controller in fact does **not support** the (the newer version) replacement operation and **it stops execution entirely**.

I will test it with the default hardware controller, thanks for your advice.

closed 2021-08-31 by petrvancjr