justagist/panda_simulator #59: Errors when running roslaunch panda_gazebo panda_world.launch

ChaoticHans, 2021-11-24:
Starting roslaunch panda_gazebo panda_world.launch makes the program run gazebo, but there are still errors showing up:

[gripper_action_server_emulator-13] process has died [pid 1825, exit code 1, cmd /home/vm-ros/catkin_ws/src/panda_simulator/panda_sim_custom_action_server/scripts/start_gripper_action_server.py __name:=gripper_action_server_emulator __log:=/home/vm-ros/.ros/log/1679ce04-4d30-11ec-81fb-000c299c877b/gripper_action_server_emulator-13.log].
log file: /home/vm-ros/.ros/log/1679ce04-4d30-11ec-81fb-000c299c877b/gripper_action_server_emulator-13*.log

[joint_trajectory_server_emulator-12] process has died [pid 1818, exit code 1, cmd /home/vm-ros/catkin_ws/src/panda_simulator/panda_sim_custom_action_server/scripts/start_joint_trajectory_server.py __name:=joint_trajectory_server_emulator __log:=/home/vm-ros/.ros/log/1679ce04-4d30-11ec-81fb-000c299c877b/joint_trajectory_server_emulator-12.log].
log file: /home/vm-ros/.ros/log/1679ce04-4d30-11ec-81fb-000c299c877b/joint_trajectory_server_emulator-12*.log

cchristofi, 2023-05-08:
I know it's been a long time, but have you found any solution to that?

illusionsoftruth, 2024-09-19:
I believe it has something to do with numpy version. There should be other error messaged saying "module compiled against API version 0x10 but this version of numpy is 0xd" or "AttributeError: module 'numpy' has no attribute 'typeDict' ". This is because "np.typeDict" is a deprecated alias for "np.sctypeDict" and has been so for over 14 years
([6689502](https://github.com/numpy/numpy/commit/668950285c407593a368336ff2e737c5da84af7d)).
A deprecation warning will now be issued whenever getting np.typeDict.
([gh-17586](https://github.com/numpy/numpy/pull/17586))

I solved this error setting my numpy version to 0.23.