justagist/panda_simulator #65: Cannot do path planning with Move It Demo

jjpatino-byte, 2022-09-14:
Hi, I am trying to do the path planning with the demo launch file.  I get this output:
[ INFO] [1663191613.654849329, 106.622000000]: Loading robot model 'panda'...
[ WARN] [1663191613.657428198, 106.622000000]: Link 'panda_link0_sc' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1663191613.657568404, 106.622000000]: Link 'panda_link1_sc' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1663191613.657605370, 106.622000000]: Link 'panda_link2_sc' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1663191613.657681129, 106.622000000]: Link 'panda_link3_sc' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1663191613.657838645, 106.622000000]: Link 'panda_link4_sc' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1663191613.657897496, 106.622000000]: Link 'panda_link5_sc' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1663191613.657952608, 106.622000000]: Link 'panda_link6_sc' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1663191613.657980712, 106.622000000]: Link 'panda_link7_sc' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1663191613.658007871, 106.622000000]: Link 'panda_link1_sc' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1663191613.658031014, 106.622000000]: Link 'panda_link1_sc' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1663191613.658053110, 106.622000000]: Link 'panda_link1_sc' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1663191613.658077937, 106.622000000]: Link 'panda_link1_sc' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1663191613.658136617, 106.622000000]: Link 'panda_link0_sc' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1663191613.658172727, 106.622000000]: Link 'panda_link1_sc' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1663191613.658197654, 106.622000000]: Link 'panda_link2_sc' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1663191613.658227046, 106.622000000]: Link 'panda_link3_sc' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1663191613.658256814, 106.622000000]: Link 'panda_link4_sc' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1663191613.658285944, 106.622000000]: Link 'panda_link5_sc' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1663191613.658311058, 106.622000000]: Link 'panda_link6_sc' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1663191613.658332689, 106.622000000]: Link 'panda_link7_sc' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1663191613.658354568, 106.622000000]: Link 'panda_link2_sc' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1663191613.658377774, 106.622000000]: Link 'panda_link2_sc' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1663191613.658400616, 106.622000000]: Link 'panda_link2_sc' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1663191613.658434332, 106.622000000]: Link 'panda_link0_sc' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1663191613.658457255, 106.622000000]: Link 'panda_link1_sc' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1663191613.658478167, 106.622000000]: Link 'panda_link2_sc' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1663191613.658499219, 106.622000000]: Link 'panda_link3_sc' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1663191613.658520922, 106.622000000]: Link 'panda_link4_sc' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1663191613.658543771, 106.622000000]: Link 'panda_link5_sc' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1663191613.658565630, 106.622000000]: Link 'panda_link6_sc' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1663191613.658587961, 106.622000000]: Link 'panda_link7_sc' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1663191613.658610137, 106.622000000]: Link 'panda_link3_sc' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1663191613.658634756, 106.622000000]: Link 'panda_link3_sc' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1663191613.658662992, 106.622000000]: Link 'panda_link3_sc' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1663191613.658685258, 106.622000000]: Link 'panda_link3_sc' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1663191613.658734888, 106.622000000]: Link 'panda_link0_sc' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1663191613.658760896, 106.622000000]: Link 'panda_link1_sc' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1663191613.658784857, 106.622000000]: Link 'panda_link2_sc' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1663191613.658808040, 106.622000000]: Link 'panda_link3_sc' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1663191613.658832152, 106.622000000]: Link 'panda_link4_sc' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1663191613.658854205, 106.622000000]: Link 'panda_link5_sc' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1663191613.658876397, 106.622000000]: Link 'panda_link6_sc' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1663191613.658898263, 106.622000000]: Link 'panda_link7_sc' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1663191613.658928569, 106.622000000]: Link 'panda_link4_sc' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1663191613.658951471, 106.622000000]: Link 'panda_link4_sc' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1663191613.658973652, 106.622000000]: Link 'panda_link4_sc' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1663191613.658995471, 106.622000000]: Link 'panda_link4_sc' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1663191613.659027643, 106.622000000]: Link 'panda_link0_sc' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1663191613.659050637, 106.622000000]: Link 'panda_link1_sc' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1663191613.659091766, 106.622000000]: Link 'panda_link2_sc' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1663191613.659121684, 106.622000000]: Link 'panda_link3_sc' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1663191613.659144700, 106.622000000]: Link 'panda_link4_sc' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1663191613.659178365, 106.622000000]: Link 'panda_link5_sc' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1663191613.659441123, 106.622000000]: Link 'panda_link6_sc' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1663191613.659477547, 106.622000000]: Link 'panda_link7_sc' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1663191613.659510815, 106.622000000]: Link 'panda_link5_sc' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1663191613.659540030, 106.622000000]: Link 'panda_link5_sc' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1663191613.659575395, 106.622000000]: Link 'panda_link5_sc' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1663191613.659616305, 106.622000000]: Link 'panda_link0_sc' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1663191613.659644573, 106.622000000]: Link 'panda_link1_sc' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1663191613.659676027, 106.622000000]: Link 'panda_link2_sc' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1663191613.659704674, 106.622000000]: Link 'panda_link3_sc' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1663191613.659732381, 106.622000000]: Link 'panda_link4_sc' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1663191613.659765419, 106.622000000]: Link 'panda_link5_sc' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1663191613.661300142, 106.622000000]: Link 'panda_link6_sc' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1663191613.661562281, 106.622000000]: Link 'panda_link7_sc' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1663191613.661600256, 106.622000000]: Link 'panda_link6_sc' is not known to URDF. Cannot disable/enable collisons.
[... cut ...] _group or called psm.providePlanningSceneService()?
[ INFO] [1663191620.095191767, 111.691000000]: Failed to call service get_planning_scene, have you launched move_group or called psm.providePlanningSceneService()?
[ INFO] [1663191620.095406646, 111.691000000]: Constructing new MoveGroup connection for group 'panda_arm' in namespace ''
[ERROR] [1663191650.125936324, 135.569000000]: Unable to connect to move_group action server 'move_group' within allotted time (30s)

Can someone give me a hand?

jiyooonp, 2023-04-14:
did you solve this problem by any chance?

jjpatino-byte, 2023-04-14:
nope

andrei-calin-dragomir, 2023-04-26:
If you are using the noetic-devel branch, try reinstalling the simulator as follows:

1. clone the repo of the simulator: 
> git clone -b noetic-devel https://github.com/justagist/panda_simulator
2. open the following file:
> <catkin_ws>/src/panda_simulator/dependencies.rosinstall
3. on the "panda-moveit-config" version, replace _melodic_ with _kinetic_
4. build the panda-simulator as explained in the README of the repo

After that try running the demo as explained in the README.

**NOTE** Make sure you run the server in a separate terminal

1358467720, 2023-06-06:
> If you are using the noetic-devel branch, try reinstalling the simulator as follows:
>
> 1. clone the repo of the simulator:
>
>
> 2. open the following file:
>
>
> 3. on the "panda-moveit-config" version, [...]

awesome!